foostan/crkbd #175: JLCPCB preview not showing proper cutouts on Cherry board.

cam443, 2022-06-06:
I made some modifications to the board in KiCad and exported it as a gerber for JLCPCB. This is the preview image I get:
[image]

I also uploaded the corne-cherry-v3.0.1 gerber zip from the releases section and it has the same issue. Is this just an issue with the display of JLCPCB or an issue with the Edge layer of this specific board?

keyboard-magpie, 2022-06-06:
This is quite normal, in my experience, and just a quirk of the JLCPCB gerber viewer.

foostan, 2022-06-07:
As keyboard-magpie said, this is normal at JLCPCB.

closed 2022-06-07 by foostan