foostan/crkbd #180: Ergogen YAML to match Corne pinouts

mankoff, 2022-08-25:
I thought it might be useful for Ergogen users to have a place to start that matches the Corne pins - because Corne seems to be one of the most commonly supported keyboards in various firmwares. If people can match Corne pins, they can more easily use their custom boards with different MCUs. Here is a YAML that I think in Corne-compliant re wiring.

```yaml
units:
  kx: cx
  ky: cy
  px: 2
  py: 2
points:
  key:
    footprints:
      choc:
        type: choc
        params:
          hotswap: false
          reverse: true
          keycaps: true
        nets:
          # from: P1
          # to: P2
          from: =colrow
          to: =column_net
      diode:
        type: diode
        nets:
          # from: P3
          # to: P4
          from: =colrow
          to: =row_net
        anchor:
          shift: [0, -5]
  zones:
    matrix:
      key.padding: ky
      columns:
        outer:
          spread: kx
          key:
            column_net: P21
        pinky:
          spread: kx
          key:
            column_net: P20
        ring:
          spread: kx
          key:
            column_net: P19
        middle:
          spread: kx
          key:
            column_net: P18
        index:
          spread: kx
          key:
            column_net: P15
        inner:
          spread: kx
          key:
            column_net: P14
      rows:
        bottom:
          row_net: P6
        home:
          row_net: P5
        top:
          row_net: P4
    thumbfan:
      anchor:
        ref: matrix_middle_bottom
        shift: [0, -ky]
      columns:
        inner:
          spread: kx
          key:
            column_net: P18
        home:
          spread: kx
          key:
            column_net: P15
        outer:
          spread: kx
          key:
            column_net: P14
      rows:
        thumb:
          row_net: P7
outlines:
  exports:
    patch_fingers:
      - type: polygon
        points:
          - ref: matrix_outer_top
            shift: [-0.5 kx - px, 0.5 ky + py]
          - ref: matrix_inner_top
            shift: [0.5 kx + px, 0.5 ky + py]
          - ref: matrix_inner_bottom
            shift: [0.5 kx + px, -0.5 ky - py]
          - ref: matrix_outer_bottom
            shift: [-0.5 kx - px, -0.5 ky - py]
          - ref: matrix_outer_top
            shift: [-0.5 kx - px, 0.5 ky + py]
    patch_thumbs:
      - type: polygon
        points:
          # start at bottom left (thumbs). Include MCU.
          - ref: thumbfan_inner_thumb
            shift: [-0.5 kx - px, -0.5 ky - py]
          - ref: thumbfan_inner_thumb
            shift: [-0.5 kx - px, 0.5 ky + py]
          - ref: matrix_inner_top
            shift: [-0.5 kx - px, 0.5 ky + py]
          - ref: matrix_inner_top
            shift: [0.5 kx + px + 30, 0.5 ky + py]
          - ref: matrix_inner_bottom
            shift: [0.5 kx + px + 30, -0.5 ky - py]
          - ref: matrix_inner_bottom
            shift: [0.5 kx + px + 30, -1.5 ky - py]
    # controller_area:
    #   main:
    #     type: polygon
    #     points:
    #       - ref: matrix_index_top
    #         shift: [0, 0.5 ky + py]
    #       - ref: matrix_index_top
    #         shift: [53 - px, 0.5 ky + py]
    #       - ref: matrix_index_top
    #         shift: [53, 0.5 ky]
    #       - ref: matrix_index_bottom
    #         shift: [53, -1 ky]
    #       - ref: thumbfan_outer_thumb
    #         shift: [-0.75 ky - py, 0.5 kx + px]
    #       - ref: matrix_index_bottom
    #         shift: [0, -ky]
    switches:
      main:
        type: keys
        side: left
        size: 14
        bound: false
    keycaps:
      main:
        type: keys
        side: left
        size: [kx, ky]
        bound: false
        corner: 0
    keywell:
      patch_fingers:
        type: outline
        name: patch_fingers
        operation: add
      patch_thumbs:
        type: outline
        name: patch_thumbs
        operation: add
    base_outline:
      main:
        type: outline
        name: keywell
      # controller:
      #   type: outline
      #   name: controller_area
      #   operation: add
    pcb:
      main:
        type: outline
        name: base_outline
pcbs:
  corne:
    outlines:
      edge:
        outline: pcb
        layer: Edge.Cuts
      keycaps:
        outline: keycaps
        layer: Eco1.User
    footprints:
      mcu:
        type: promicro
        params:
          orientation: down
        anchor:
          ref:
            - matrix_inner_top
          shift: [22, -4]
          rotate: 270
      reset:
        type: button
        anchor:
          ref:
            - matrix_inner_bottom
          shift: [kx, 0.5 ky]
        nets:
          from: GND
          to: RST
      trrs:
        type: trrs
        anchor:
          ref:
            - matrix_inner_bottom
          shift: [34, -4]
          rotate: 270
        params:
          reverse: true
          symmetric: true
        nets:
          A: VCC
          B: VCC
          C: GND
          D: P0
```

mankoff, 2022-08-25:
When run through Ergogen, it produces something that looks like this:

[image]

foostan, 2022-11-22:
I didn't see any particular advantage, so I close it.

closed 2022-11-22 by foostan