foostan/crkbd #184: Right part from crkbd wont work anymore.

Mariotpan, 2022-12-14:
Hi I was using my Corne Keyboard and it worked nicely, but after I tried to change some funcionalities on the software and flash it back to crkbd the right hand part stopped working.
The leds still turn on when I plug it into my computer but nothing happens when I press any keys from the right part.
I tried to flash a default software from VIA but the right part wont work anymore.
I have no clue of what can I have done wrong or how can I fix it, idk how could I maybe reset it or install a default and working flash
Help

cowboyliao, 2023-02-08:
I am making a corne-chocolate now, had done soldering chips work, then, I test the key by short with via website. 
Here is issue I have, the left hand is typec, and the right hand is mircousb. But I found that when I connet the left, typec can be used, but the right hand did not respond. When I connet right, both sides can be used, but the left and right key positions are reversed.
I build the firmware myself defaultly, using Atmega32U4. 
I thought it's the issuse by bootloader? but seller say he do the bootloader before.

What should I do?

cowboyliao, 2023-02-09:
I move to windows and flash the firmware by qmk toolbox with the default firmware you give, still the issue:
plug left, left work, right not work. plug right, left and right both work, but reversed. 
how could I fix it?

waffle87, 2023-02-18:
The issue of right half not working when USB is connected to left, but both halves working when USB is connected to right is very strange. But the halves being reversed when the USB is connected to the right is expected, as the firmware expects the USB to be connected to the left half by default. You can change this by [setting handedness](https://docs.qmk.fm/#/feature_split_keyboard?id=setting-handedness) appropriately, then recompiling the firmware and reflashing both halves.

Aristolon, 2023-05-06:
I ran into a similar problem. Every time the keyboard works fine after plugging the typec back in, but after a while the right part doesn't work (the left hand is the main hand by default).

SozinovD, 2023-08-19:
This line should be added to `config.h` file
```
#define SPLIT_USB_DETECT
```
P.S. It took about 3 hours for me to figure this out. Please fix it in main repo
This link helped me a lot https://github.com/foostan/crkbd/issues/170#issuecomment-1104831554

enorfelt, 2023-08-31:
> This line should be added to `config.h` file
> 
> ```
> #define SPLIT_USB_DETECT
> ```
> 
> P.S. It took about 3 hours for me to figure this out. Please fix it in main repo This link helped me a lot [#170 (comment)](https://github.com/foostan/crkbd/issues/170#issuecomment-1104831554) In my case it works with this firmware that i compiled [crkbd_rev1_via.zip](https://github.com/foostan/crkbd/files/12386699/crkbd_rev1_via.zip)

Hade the same issue. This worked for me! Thanks m8!

carl-ranson, 2023-10-06:
I'm having the same problem even with the firmware that @SozinovD  posted (as well as all the others I tried)
I'm thinking it might be a hardware fault in my case. 

Can anyone shed some light on how the controllers know which side of the keyboard they're on? Its the same firmware on both, so either it's a master/slave setup or there's a pin on the mcu that is tied high/low or something.

foostan, 2023-11-11:
The board with the USB plugged in is master.

BrassBaron, 2023-11-26:
I have the same problem, and putting the line "#define SPLIT_USB_DETECT" in config.h does not help. 
Catch is, that both halves worked and after making some changes in keymap (we all know, that this is a never ending evolution... ;) ), right half does not work anymore.  Separately they both behave like left half, that is if I change USB from one half to another, just that the right half is mirrored. I even flashed micro with some Arduino program, to make sure to erase old contents.  I changed the TRSS cable, did not help either. I guess there is some catch in communication protocol between both halves?
Any ideas?

foostan, 2024-03-15:
I'll close it because this issue is not related to design. If you have any concerns please continue at Discord https://discord.gg/aWCZWnS

closed 2024-03-15 by foostan