foostan/crkbd #186: Support for Cherry MX Low Profile Switches?

Lokei1970, 2023-01-03:
I read the documentations but didn't find the specific details about whether any of the models support Cherry MX Low Profile Switches:
https://www.cherrymx.de/en/cherry-mx/mx-original/mx-speed-silver.html
https://www.cherrymx.de/en/cherry-mx/mx-low-profile/mx-low-profile-red.html

Does any of the models support these types of switches?

keyboard-magpie, 2023-01-03:
No. The latest revision of Corne Light supports Kailh Choc v2 which are the next closest thing to Cherry MXLP.

Lokei1970, 2023-01-03:
@keyboard-magpie Thanks for the reply bro!

closed 2023-01-03 by Lokei1970

adaviloper, 2023-06-01:
> No. The latest revision of Corne Light supports Kailh Choc v2 which are the next closest thing to Cherry MXLP.

Is the Corne Light or any of the other Kailh Choc boards set with MX spacing? I couldn't find if this was the case.
As in, are the Choc PCBs the exact same physical dimensions as the MX PCBs, except the footprints for the switches are replaced with Choc footprints?
I want to build a new low-profile Corne, but I prefer MX spacing.