foostan/crkbd #198: Corne cherry v3 into mini

driccio, 2023-06-17:
Hi,

I have pcbs for corne cherry v3. I wonder if the pcb can be cut to have only 5 columns as nothing help to cut it (unlike the corne choco for instance).

Thanks for your answer.

Damien

foostan, 2023-06-25:
v3 doesn't support the mini version.

closed 2023-06-25 by foostan

driccio, 2023-06-25:
Hi foostan, 

Thanks for your answer. Even if not supported with pre-cut pcb, I thought it was possible to cut it with precautions in order to have a mini version. I understand by your answer there is no way.

driccio, 2023-06-28:
Hi,

To answer myself, here is how I cut the PCB of a cherry v3 in order to have a mini version.

**Disclaimer** : I am not an expert and I just share what I did in order to help.

**What you need to know before ?**
If you don't have LEDs, then you have nothing else to do than cutting the PCB.
If you have LEDs, then you will break the LEDs circuit on the right hand. But it is easy to solve.

**Where should we cut the PCB ?**
Please see purple lines 1 and 2 that materialize the cutting line. 

[image]

**How to cut the PCB ?**
Take a cutter and begin slowly to mark the line on both side of the PCB. Repeat operation in order to have deep lines that would allow you to break the PCB. Be careful !

[image]

**How to fix right hand ?**
I repair the break by connecting 2 LED with a piece of cable as you can see here :

[image]

Result :-)

[image]

Enjoy

Damien