foostan/crkbd #210: using pi pico instead of pro micro controller possible?

pathnirvana, 2024-02-06:
This is a question - not an issue.. Not sure where to ask it. Sorry if it is not the correct forum.

Is it possible to use the pi pico mcu based pro micro form factor controllers instead of the atmega pro micro. I know the mcu operating voltage is different. Has anyone used it with the pcb here?

eg controller here https://www.aliexpress.us/item/3256805943704472.html

keyboard-magpie, 2024-02-06:
Yes- with the use of the QMK converter tools: https://docs.qmk.fm/#/feature_converters?id=supported-converters

The particular board you link falls under "rp2040_ce"

pathnirvana, 2024-02-10:
Thank you @keyboard-magpie that was very useful

closed 2024-02-14 by foostan

pathnirvana, 2024-03-24:
for anyone who is interested here is the repo for corne+vial using promicro rp2040 from aliexpress
https://github.com/pathnirvana/vial-qmk/tree/vial/keyboards/crkbd/keymaps/vial_rp2040

fgeck, 2026-02-13:
I am running into the exact same question when building the v3 corne. 
How would I install the pi pico as it has 13 pins instead of 12?
@pathnirvana can you give some advice here? maybe also a short how to for using the converters, and then building the firmware?