foostan/crkbd #213: four switches on the pcb of the left hand not working

OneCrazyman, 2024-03-06:
I am making a keyboard called corne v3 cherry.

After attaching diode and produino, short circuit all switch parts directly. Left and right both work well, but the four switches on the pcb of the left hand not working. When seen with the naked eye, all are soldered.

[image]
[image]

OneCrazyman, 2024-03-06:
[image]

foostan, 2024-03-08:
The problem with one vertical row not moving is in the connection with ProMicro. Try re-soldering it cleanly.

foostan, 2024-03-15:
I'll close it because this issue is not related to design. If you have any concerns please continue at Discord https://discord.gg/aWCZWnS

closed 2024-03-15 by foostan