foostan/crkbd #219: Schematic Cleanup + PCB stackup considerations

fabianmuehlberger, 2024-03-21:
Hi, 

Since I am looking for a new keyboard, I came across this project. 
Looking at the board design files for the Corne chocolate from the main branch, I would suggest a rearrangement of the schematic symbols to a more functional layout. 

The schematic should follow general guidelines like:
- Symbol naming/numbering from left top to right bottom, 
- Upper left Power, IO
- right side MCU, peripherals (flash, crystal)
- do not cross lines if possible 
- PWR labels always Up, GND labels always down 
- labels on symbols should sit on the line 

see example:
[image]

I have not checked if the left and right side have significant functional differences, if so, making hierarchical sheets for duplicates would be suggested too, e.g. keys, LEDs, MCU on separate sheets and references to them for left and right side if the components are the same. 

 

Regarding the PCB, I would suggest switching to a 4 layer PCB. Since ground pour is only on the top side with a high number of traces on the top and bottom, as well as layer changes of signal lines, current return paths are VERY restricted. A 4 layer PCB with 2 ground inner layers would make routing easier and improve the board overall. 

I made some explanatory changes in the corne chocolate schematic on my [fork](https://github.com/fabianmuehlberger/crkbd)
Please let me know if you are interested in a redesign, if there is interest, I am happy to contribute.

Best Fabian

foostan, 2024-03-21:
Thank you for your suggestion. Please provide evidence to evaluate the validity of your proposal. thank you.

fabianmuehlberger, 2024-03-21:
What do you mean with validity? PCB design is mostly based on guild lines rather than fixed rules. 
The parts I mentioned are pretty basic and would apply to most simple PCBs with one populated side. 

I found [this guide](https://www.pcbway.com/blog/PCB_Design_Tutorial/How_to_make_clear_and_well_organized_schematics.html) with basic recommendations for the schematic. 

[Elegant schematics](https://resources.altium.com/p/creating-elegant-and-readable-schematics)
[schematic checklist](https://resources.altium.com/p/schematic-review-checklist)

A best practices regarding ground can be found [here](https://resources.altium.com/p/pcb-ground-plane-best-practices-your-multilayer-stackup)

foostan, 2024-03-21:
Thank you for sharing. I'll confirm the guide and fix them.

About 4 layer, could you tell me your opinion more. Could you share Pros/Cons?

fabianmuehlberger, 2024-03-22:
Hey, It's not about the guide, these enhancements can be separately applied. 

1. Schematic rearrangement/refactor is to make it more concise. The goal is to make it easier to read and debug if there are problems.  Following basic guidelines will help adapt and improve the design in the future and make it more portable. 

2. PCB stackup is a big topic itself, and I am not an expert. But 4 layer PCBs are the norm nowadays and have gotten cheap to produce; therefore we can get the advantages over a 2 layer board with basically no downsides. I see no reason for not using a 4 layer board 
These are:
- Inner ground layers: generally speaking, you want all traces separated by a ground layer. This helps with **crosstalk** and gives provides a solid **current return path**. With 2 layers, this is only possible when using one side for traces and the other for ground. Obviously, this can only be done on low density boards  (low component count; space for routing). 
- Easier Routing: With a 4 layer board, it is much simpler since you can use the 2 inner layers for GND and the PWR bus and use the outer layers for the signal traces. This makes routing easier, since you simply have more space to do so. 
- EMI: For commercial products in the EU an EMI/EMC test is mandatory. So far, I have not produced a PCB for the market, but it's a big topic. Following the industry norm wherever you can helps comply with those standards. 

Best

foostan, 2024-03-28:
Thank you for sharing. I'll read the docs you shared.

fabianmuehlberger, 2024-03-30:
From your answer, I assume that you are not interested in me contributing to the project. 
I close this issue then.

closed 2024-03-30 by fabianmuehlberger

fabianmuehlberger, 2024-11-18:
Since there is an ongoing discussion regarding EMI problems, [#265 ](https://github.com/foostan/crkbd/issues/265#issuecomment-2480519959)  I decided to reopen this issue. 
It seems that the concerns in this issue are directly related with the current issues of the hardware.

foostan, 2025-02-09:
I'm currently prototyping with 2 layers at https://github.com/foostan/crkbd/tree/v4_1_1. If this proves stable enough, It'll continue with 2 layers. If this doesn't work, I may redesign with 4 layers as 4.1.2.

fabianmuehlberger, 2025-02-09:
Hey @foostan great to see you are making progress! 

I will take a look at your design, maybe I can help improve it here and there. 
One thing that I noticed first was the VLED and LED traces intersecting the board south->north and running parallel to the USB lines. There are 2 things to note here. I highlighted those traces. 
I took a look at the LEFT board, but the same principle applies to the right side

[image]

1. Crosstalk: High-speed signal lines can have crosstalk on parallel signal traces. Since you have some USB issues, you might want to reduce the chance as much as possible.
2. Current return path. Since your USB Connection and therefore your GND is in the upper-right corner (left board), you want a current return path from your IC to that area, that is not intersected by traces. 

VLED: you could try routing the VLED from the down and left. Making a circle clockwise around the board instead of cutting through the middle,
LED: I am not sure if the order of attached LEDs is important: if not, you could do the same for the LED line, instead of routing top to bottom, you could try to attach the LEDs from the left side.

This would give you a chance to have a solid copper GND from the IC to the USB port, as well as a solid ground layer under the USB lines. Which should help with the EMI issues you are experiencing. 

Best

foostan, 2025-02-09:
@fabianmuehlberger Thank you for your suggestion! I'll check later and update it 👍 Thank you!

foostan, 2025-02-11:
@fabianmuehlberger 

> This would give you a chance to have a solid copper GND from the IC to the USB port, as well as a solid ground layer under the USB lines. Which should help with the EMI issues you are experiencing.

Is the mean of the VLED and LED improvement proposal pointed out above? I understand it is not good that these lines are in the center of the board and separate GND?

foostan, 2025-02-11:
It's difficult to solve this issue by 2 layer for me. I may use 4 layer for it.

fabianmuehlberger, 2025-02-15:
The reason for the proposed improvements are, as mentioned, for a good current return path. You can read about this in this article.  https://www.protoexpress.com/blog/current-return-path-signal-integrity/ or if you prefer a video https://www.youtube.com/watch?v=lv8CZSc0o-c&t=180s. A lot of what I know is from the altium academy. 

To sum it up: Every signal has some (maybe small) current. It follows a path from the power supply to components and then to the ground plane or layer. Some return current will be induced into the ground plane as displacement current!
 Every signal that is not a simple DC signal with some frequency is an RLC circuit (it has a resistance, capacitance and inductance part to it). You can imagine it as a cupping between the signal line and the ground.  This determines the return path of the return current. With high-frequency signals, the return current tends to follow the signal line. 

Very loosely spoken: When you "cut" the ground layer with other traces holes, vias etc., you are making it harder for the return current to follow its desired path. 

Regarding the USB lines: The second reason for a solid ground is, that not only the coupling between the USB and the GND layer is better. But it also shields the lines from other signals. Your LEDs are also a high-speed signal that can couple with the USB lines and introduce noise.

I highly recommend watching this video https://www.youtube.com/watch?v=VRJI0X-6yTg&t=120s it is an excellent primer for the design of High-speed signals.

foostan, 2026-11-14:
Thank you for the suggestions, and sorry for the slow progress. In v5 (#328):

- The schematic has been restructured to share the left/right sheets
- The PCB has been fully re-routed, including USB and power
- The board remains 2-layer for now

The v5 boards are currently being tested for EMI/stability (see #265), and I expect to reach a conclusion in a few weeks. If 2 layers turns out to be insufficient, a 4-layer stackup will be the next option to consider.

Feedback on the v5 schematic and layout is very welcome.