foostan/crkbd #220: Corne Cherry v2 detailed manual needed

set-st, 2024-03-22:
Is any v2 flashing manual exist? I trying found instructions about enabling oled displays. Of course im add `#define USE_I2C` and separate halfs is working well, but then i try connect it by wire and screens not works, and on right side nothing works exept backlight. And will be exelent to be got some details about resistors and jumper to data pin which detached by default.

foostan, 2024-03-28:
- Where did you get the PCB? 
- Did you build firmware yourself? (Normally you don't need to change anything) 
- Is the cables broken?

set-st, 2024-04-12:
I found short cut on one of i2c pin and data, so it can't work properly. Now i order official pcb.

foostan, 2024-08-04:
It will be closed as there have been no comments for some time. Reopen if necessary.

closed 2024-08-04 by foostan