foostan/crkbd #221: Add indicator LEDs

jradtilbrook, 2024-03-26:
Can I first just say, the recent v4 release is absolutely awesome. I've been using a corne for about 6 months now and love it! I came here looking to build my own one from scratch and see about a few mods. I love that the MCU is onboard now and that there is the option of 4 more switches. It's exactly what I wanted - just slightly more keys to reduce the layers/tap-dance/etc 👏 

I think the only thing holding this back from being the perfect keyboard for me is the addition of some LED indicators. Then there is little need for a display and we can add 4 extra switches 🙌 

Is that something you would be open to a PR with?

jradtilbrook, 2024-03-26:
Some extra info that would be useful:
- I suppose adding a mcp23018 to one or both sides (so it allows for left/right master)
  - this is similar to how other keyboards do it
- add a cutout to the top plate so they are visible. I haven't looked closely but perhaps above the outermost thumb switch

foostan, 2024-03-28:
Thank you for your suggestion. 

However, I don't plan on adding an indicator because I want to keep the design as simple as possible. 

I recommend you fork it and proceed with it as your own project.

jradtilbrook, 2024-03-29:
Thanks for your response. Okay I'll do that 👍

Could you explain a little how to set up kicad for modifying it? Do I need to use the `./pcbs/common` folder to import into kicad somewhere for extra footprints?

Otherwise, how can I start with trying to modify the corne cherry?

jradtilbrook, 2024-03-29:
Oh and another question @foostan; I've looked at the schematic already for the corne and it looks like each switch is wired directly to a RP2040 pin. I thought most keyboards use a matrix for the switches so that they didn't have to use so many IO pins from the MCU. Is my finding correct or am I misunderstanding the schematic?

And if that is correct, are there no more spare pins on the RP2040 to use for indicator LEDs?

jradtilbrook, 2024-03-29:
Okay doing a bit more research it does appear to use "direct wiring"

foostan, 2024-03-30:
> And if that is correct, are there no more spare pins on the RP2040 to use for indicator LEDs?

Ah, yes.

closed 2024-03-30 by jradtilbrook