foostan/crkbd #223: V4 Cherry build question

jaeheonji, 2024-03-30:
Hi, I received the following request from JLCPCB while ordering the v4 version (cherry), but I do not know anything about it, so I am asking a question.

[image]

Above image, the indicated pad and trace is connected. Should I need to separate them?

foostan, 2024-03-30:
It's no problem. 
I think it would be best for you to confirm the data with kicad before ordering to the factory.
[image]

foostan, 2024-03-30:
I have fixed this just in case. Thanks

closed 2024-03-30 by foostan

jaeheonji, 2024-03-30:
Thank you so much @foostan !