foostan/crkbd #229: Hardware Issue: Problematic routing of USB differential pair!

fabianmuehlberger, 2024-04-06:
Hi, 
I took a second look at your PCB, and I highly recommend revisiting the USB routing. USB D+ and D- should always be the highest priority with the following criteria: 

- short traces 
- correct impedance
- matched length 
- consider ground/ via fencing. 

Not following these standards could lead to unreliable or faulty behavior of the keyboard.

bjsemrad, 2024-05-08:
I'll add this here since that wake change I haven't had an issue with a direct usb connection to the laptop. However through my level 1 KVM I still experience this issue, but appears to be only on my work mac and not my Linux pc.

I will keep debugging but leaving this here since it could be related to the usb spec constraint pointed out.

foostan, 2024-05-08:
Thank you for sharing. I keep to investigation too.

foostan, 2026-11-14:
Thank you for pointing this out. In v5 (#328), the USB routing has been fully redesigned:

- New MCU (STM32G0B1) placed close to the USB-C connector, with short D+/D- traces (about 17 mm to the host connector)
- ESD protection (USBLC6-2P6) on each USB-C port
- The halves are now connected with USB-C instead of TRRS

Since the MCU runs USB Full Speed (12 Mbps), the remaining small length mismatch is well within tolerance. The v5 boards are currently being tested, and I expect results in a few weeks.

I'll close this issue once v5 is released. If you have time, I'd appreciate a review of the v5 layout.