foostan/crkbd #231: How to order? / Wrong parts placement when ordering from jlcpcb

MaxWinterstein, 2024-04-08:
Love your work, and currently typing this on an old corne <3

I tried to upload the files to jlcpcb but somehow the parts are mostly wrong rotated or misplaced. This happens for both cherry as well as choc. E.g. USB port is floating.

[image]

[image]

There are others on Reddit having this issue too: https://www.reddit.com/r/crkbd/comments/1budyel/comment/kxuc1wx/

Thanks!

foostan, 2024-04-08:
I answered the same question on my server 
https://discord.com/channels/715899960327012423/742781449731702824/1225199804477472858

> I don't know the details, but it may be a compatibility issue between the jlcpcb system and the data generated by kicad. The jlcpcb engineers will check and adjust the data, so there is no problem with the way it is.

MaxWinterstein, 2024-04-08:
Thanks, will try to get them to adjust the data.

Sadly, the headphone jack is not in stock, probplaby since a few days already. Do you know any compatible alternative?

foostan, 2024-04-10:
There is no alternative. If you want to change, you need to change the footprint. I'm waiting to be stocked.

RodolfoBar, 2024-04-13:
In my case the engineer is asking me to confirm the placement of the parts. So I was thinking, would it be easier to have a diagram in the documentation/build guide that has the correct placement of the parts so that you can maybe avoid these questions in the future?

RodolfoBar, 2024-04-13:
I'm having a similar issue with JLCPCB. I was wondering if there was any way that I could confirm the orientation and placement of the parts. I realise this might be a bit of a request, however the images that are on the README.md in the root are quite low resolution and I don't have that much knowledge with circuitry to really understand it. Thank you!

Xartos, 2024-04-20:
I'm currently looking at creating a fork that uses [PJ-606-6A](https://jlcpcb.com/partdetail/Korean_HropartsElec-PJ_6066A/C145806) as the audio jack instead. However since @foostan wrote that there's no alternative I'm wondering if there's anything that I'm missing with it that makes it incompatible, of if that was meant that there's no alternative with the same pinout. Because other than the pinout they seem to have similar footprint so only minimal changes to the pcb is required.

foostan, 2024-05-06:
I released https://github.com/foostan/crkbd/releases/tag/v4.1 for the connector issue.

closed 2024-05-06 by foostan

MaxWinterstein, 2024-05-27:
Thanks, I was able to order 4.1 from jlcpcb. They placed all parts correctly - but IIRC I checked and payed all boxes that involved production file verification and confirmation.

[image]

New connector works fine, currently typing on a v4.1. Thanks for the awesome work!

[image]

Just had do use white switch holders, the black ones where out of stock at that moment.

kijewskimateusz, 2024-06-03:
Thanks for great discussion there. When ordering my pcb, I have also been asked whether polarities of the components C5149201 are correct or not. (SMD-LED)

Could you please help me understanding if it is correctly applied by the JLCPCB team? Bellow I'm attaching render shared by the team. 
[image]

MaxWinterstein, 2024-06-03:
@kijewskimateusz 

Attached the file I accepted for comparison.

[image]

Even I made MX style, and yours is choc, it can be compared. By looking at the LEDs it looks good to me, as the little triangle needs to be in the marked corner. The real LEDs have some little cutout.

[image]

giograno, 2024-06-04:
Hei @MaxWinterstein, I am also trying to order the v4.1 with jlcpcb. The sockets and the trrs connectors are rotated/misplaced. As far as I understood, in your case jlcpcb itself managed to correct them, right?

mhanberg, 2024-06-12:
@kijewskimateusz I am facing a similar issue with JLCPCB.

My switches in the mockup they sent look like yours, but i'm wondering, are they upside down? Did you end up getting assembled board and confirmed that that is correct?