foostan/crkbd #234: Where are the rest of the board designs?

Koltsz, 2024-04-26:
Hi,

how can i get a hold of Corne Light v2 Low edition PCB? I can't seem to find it anywhere, can only find choc and mx cheery ones

SparrowMike, 2024-04-26:
Not sure what exactly you mean.

But if you're after just the pcb here is one 

https://www.etsy.com/sg-en/listing/1113750577/corne-light-v2-pcb?ga_order=most_relevant&ga_search_type=all&ga_view_type=gallery&ga_search_query=corner+light&ref=sr_gallery-1-2&content_source=c3da52396f7fdb4c7e623868bcf19554bf095be1%253A1113750577&organic_search_click=1

Koltsz, 2024-04-26:
> Not sure what exactly you mean.
> 
> But if you're after just the pcb here is one 
> 
> https://www.etsy.com/sg-en/listing/1113750577/corne-light-v2-pcb?ga_order=most_relevant&ga_search_type=all&ga_view_type=gallery&ga_search_query=corner+light&ref=sr_gallery-1-2&content_source=c3da52396f7fdb4c7e623868bcf19554bf095be1%253A1113750577&organic_search_click=1
> 
> You could also find ultra light / light designs here https://github.com/petejohanson/crkbd/tree/board/corne-ultralight

Thanks for this,

I'm looking for the PCB files so I can get the PCB printed myself

djw33, 2024-04-28:
There are different tags for each of the different versions. Is that what you're talking about:

[image]

foostan, 2024-05-06:
Thank you for answering. You can find it by changing other tags.

foostan, 2024-05-22:
I'll close this issue because it looks already solved.

closed 2024-05-22 by foostan