foostan/crkbd #235: jlcpcb production files for v3

kuritonasu, 2024-04-30:
May I kindly request if you could provide the bom, xml and cpl files as part of the v3 final release? I understand that it's not hard to produce them using kicad, however given the amount of options that a user can configure while producing the files, it's a bit less intimidating, and more beginner friendly if they can bypass kicad completely.

Thanks again for your awesome work and for considering my request!

mattsn0w, 2024-05-04:
JLCPCB has a howto on this exact topic. 
https://jlcpcb.com/help/article/16-How-to-generate-Gerber-and-Drill-files-in-KiCad-6

These instructions are for version 6 of kicad, but the dialog sections and options are all still the same on kicad version 8.

foostan, 2024-05-06:
Sorry, I will no longer be updating for v3 🙏

closed 2024-05-06 by foostan