foostan/crkbd #238: Corne V4.1 PCB question.

notPrivate-care, 2024-05-11:
Hey, I am a newbie, don't know anything about pcb, embedded systems, not even pressed a mechanical switch in my life. I have been wanting to use a minimal and split keyboard like Corne since a long time, but money and lack of knowledge has always stopped me. I have a few questions about Corne V4.1:

1) If I am correct, this hotswappable version allows me to order pcb, which has all the components soldered, allowing me to use the keyboard without soldering anything myself, and by just putting the switches and keycaps.

2) Why cant I upload the production files to a local pcb manufacturer? Jlcpcb requires to order at least 5 pcb, and 2 assembly service, which is really really expensive, and I only need 1 keyboard.

Thank you!

foostan, 2024-05-11:
1. Yes, you don't need solder yourself if you order as PCBA and all components will be soldered.

2. What do you mean? I uploaded files for JLCPCB as a sample. So you can order it anywhere.

notPrivate-care, 2024-05-12:
Thank you for your reply!

1. For example, I want to order one keyboard (both left and right) from https://www.pcbpower.com/ . What settings do I have to write? I have no idea.

2. Is a case and plate necessary? if so, where can I order them?

3. I think I should do the cherry mx build as they are widely available and cheaper. Is this correct?

foostan, 2024-05-13:
1. Please read the documentation of pcbpower
2. You can order at jlcpcb or others, but I am not sure about other services.
3. It's depends on you.

foostan, 2024-05-22:
I'll close this issue. Please reply if you have any concerns or questions.

closed 2024-05-22 by foostan

notPrivate-care, 2024-05-26:
> I'll close this issue. Please reply if you have any concerns or questions.

Is it necessary to use plate and case? Can I use with just pcb with rubber feet?

Still-Here-0, 2025-02-15:
Nether of those are necessary (not even the rubber feets). However they will ensure quality and a longevite. 

PS: if you do not use plates in your build its keys will be kinda wobbly