foostan/crkbd #239: microcontrollers

LucasSantos52, 2024-05-20:
Hi, I'm a keyboard enthusiast, I started looking into the subject a few months ago, and in all the customizable keyboards I find on the internet, some type of microcontroller is always soldered onto the board, however, looking at the model on JLCPCB I noticed that on the In the microcontroller region, there is no drilling for the pins, which are usually soldered.

What type of microcontrollers can I use in this version 4?

If I'm not wrong, I need to solder one to the board, right?

I looked through the zip files but wasn't able to find anything

Eingin, 2024-05-20:
The [V4 release](https://github.com/foostan/crkbd/releases/tag/v4.0) info answers your question.

The board no longer uses a promicro and instead has a soldered RP2040. This is located near the center of the board by the bottom row. 
If you want a 4v, you will most likely have to order it as a PCBA as soldering the microcontroller plus the supporting components is tricky.

foostan, 2024-05-22:
I'll close this issue because it looks already solved.

closed 2024-05-22 by foostan