foostan/crkbd #241: V4.0のファームウエア

detectivekounan, 2024-05-21:
[crkbd_rev4_0_standard_via.uf2](https://github.com/foostan/kbd_firmware/raw/main/keyboards/crkbd/qmk/qmk_firmware/.build/crkbd_rev4_0_standard_via.uf2)

こちらを書き込み後から、キーボードがうまく認識しなくなってしまいました。
BOOTスイッチを押しながらUSBケーブルを接続することで、WindowsエクスプローラにRPI-PR2は認識されます。
が、uf2ファイルを書き込んだあとから、WindowsにUSBを接続・切断の音が繰り返し流れるようになり、TRRSケーブルで接続している反対側のキーボード(左のキーボードにUSBを接続している場合は右)が全く反応しません。
復旧方法ありますでしょうか？

foostan, 2024-05-22:
原因の切り分けをしたいのでいくつか確認させてください
- ファームウェアは左右両方に書き込んでいますか？
- [以前のバージョン](https://github.com/foostan/kbd_firmware/tree/264693cbf27223810b508dfa66a41a4b3ae2f871/keyboards/crkbd/qmk/qmk_firmware/.build) のものを試して頂けますか？

detectivekounan, 2024-05-22:
@foostan 様
以前のバージョンを書き込んだら使えるように戻りました。
お手数をおかけいたしました。
ありがとうございました。

foostan, 2024-05-22:
ご確認ありがとうございます。
最新バージョンについて調査及び修正します。
ご報告ありがとうございました！

foostan, 2024-05-22:
同一のファイルで検証してみましたが特に問題は見つかりませんでした。
こちらのissueはこのままcloseさせていただきます 🙏

closed 2024-05-22 by foostan