foostan/crkbd #243: Killed the rPi by first time powering up with connected TRS cable?

MaxWinterstein, 2024-05-27:
I ordered 5 PCB pairs. 4 Work great, but 1 is missing any signs of life.

When I tested them, I accidentally powered one factory new PCB pair with a connected TRS cable. Is it possible to fry the chip when doing this? Like a brand new chip contains some firmware that has voltage where there shouldn't be any?

gpmontt, 2024-05-28:
Maybe you need to secure the TRS cable. Some diodes can be helpfully here

defekkt, 2024-06-05:
had the same issue, might have been a voltage spike when connecting TVRS cable *after* connecting USB.
turned out it killed the V-USB-> 5V Diode. might be worth checking for you too ;-)

MaxWinterstein, 2024-06-06:
@defekkt can you give me more information on that? Not sure where this diode is, and how to check it. Thx :)

defekkt, 2024-06-16:
1.grab a multimeter, put it in diode mode
2. on a known-working/good board/diode measure the diode in both directions
3. then check it on the faulty board, compare the values (should be different depending on the direction).

here is the diode, next to the usb-connector 
[image]

closed 2024-10-19 by foostan