foostan/crkbd #244: v4.1 version left part issue

vadimkorzik, 2024-05-27:
hello @foostan 
Thanks for the great project. 
I am facing a problem.
I have assembled two sets of the keyboard version 4.1 and both have the same problem with the left part. it is correctly recognized as a drive to write a firmware but doesn't work correctly as a keyboard. Windows says "USB device not recognized". Both right parts work fine. The left part connected to the right via trrs doesn't work either. Looks like it is a software issue as it starts showing some info on oled but after a second "hangs" and does not respond to key presses. Flashed via firmware supplied and tried to build it myself with via and default km. Really appreciate your help.

foostan, 2024-05-28:
It is possible to write firmware, right? After writing the firmware, it doesn't get recognized as a keyboard. I'm using the same data and it works fine.

vadimkorzik, 2024-06-03:
hi. turns out it was most likely my fault, i reflowed chips on both plates and they work now. looks like i overheated rp2040 while soldering.

closed 2024-06-03 by vadimkorzik

ShionSumeragi, 2024-10-29:
> hi. turns out it was most likely my fault, i reflowed chips on both plates and they work now. looks like i overheated rp2040 while soldering.

Can you tell what temperatures you used? I seem to have made the same mistake