foostan/crkbd #248: Corne v4: what components to buy?

mhanberg, 2024-06-07:
Hi!

I am new to building keyboards from scratch, so apologies if these are silly questions.

I have purchased some Corne v4.0 PCBs (not pre assembled, and still need to solder on the components, like keyswitch sockets, LEDs, etc.

I was under the impression that I would need to buy some surface mount diodes, but on the PCB, I'm not sure where they would go.

So, that leads me to ask the blanket question: what components are required for this build?

Thanks!

jnfengke, 2024-06-07:
No diodes needed. It uses direct wiring. Everything is listed in the BOM. The TRRS connector might be hard to source. Soldering QFN would be challenging if you aren't very experienced.

Thank you

Ke Feng

foostan, 2024-06-08:
[RP2040](https://jlcpcb.com/partdetail/RaspberryPi-Raspberry_PiRP2040/C2040) is not assembled too? If yes, you need to prepare the reflow soldering environment.

mhanberg, 2024-06-11:
Due to my inexperience with building keyboards from this level, I didn't realize that the new design wasn't really DIY friendly.

So, I went ahead and put in a PCBA order from JLCPCB.

Thanks for the help!

closed 2024-06-11 by mhanberg