foostan/crkbd #249: Corne v4.1 3x5 Chocolate with OLEDs

a-manumohan, 2024-06-08:
I am planning to build a corne with the following parameters.

- Latest version(currently 4.1) 
- 3x5 Layout
- Kailh choc switches
- **OLEDs**

I haven't seen any samples of v4 or v4.1 with OLEDs anywhere on the internet yet. The changelog from v3 to v4 mentioned continued support for OLED and I noticed the pin headers on the PCB. But I can't find a top plate GERBER with a cutout for OLED displays. 2 Given options are for mini choc corne and extended choc corne with the cutout for the additional 2 switches. 

I do not know PCB design or how to work with kicard. So my questions is

**Do you have any plans to add designs for a top plate that supports OLED?**

kijewskimateusz, 2024-06-14:
@brujoand could you please elaborate about the difficulty of soldering the OLED to the PCB? I think it may be worth opening an issue to add socket for the OLED.

brujoand, 2024-06-14:
@kijewskimateusz there are a couple of challenges, one being that I'm no expert on soldering.  

First I learned that I had to trim the legs of the OLED PCB so that it just sit's a couple of mm out from the through hole on the Corne PCB. If not then it's really difficult to get the soldering iron into an angle where it doesn't touch (and melt) the minijack. 

The other challenge is that the connecting ring you're supposed to solder the OLED PCB legs to are so thin that it's difficult to heat them properly. This causes solder to only attach to the OLED PCB leg, and this is probably where I broke my board by over heating the PCB while trying to heat the connector. 

I tried both with the stock tip and the smallest tip from my Pinecil soldering iron. Maybe someone will have better luck with more professional hardware, or better soldering skills.

foostan, 2025-02-09:
This issue looks to be resolved so I will close it.

closed 2025-02-09 by foostan