foostan/crkbd #251: replace the TRRS connectors with USB C connectors

plainoldcheese, 2024-06-26:
The TRRS connectors commonly used on split keyboards are not ideal for their purpose and the PJ-320A connectors are often difficult to find.  There is a possibility to short GND and VCC when disconnecting the TRRS cables while the keyboard is powered  (though smaller with the TX in the middle as on the current layout). USB-C works both directions and the parallel nature of the connections allow for safer connecting and disconnecting.

This is a quick proof of concept. This  USB-C connectors footprint allows for the OLED screen to stay where it is and for the location of the connector to stay the same as where the TRRS connector was. 

Using the same USB-C that is used at the top of the board would be ideal, but the footprint would overlap with the OLED unless the position of either the OLED header pads or the connector is moved. I was hesitant to move the OLED or position of the connectors as that would make current cases incompatible.

I do think the streamlined look of the dual USB-C connectors with the edge cut at the top of the board looks cool though and that would be ideal for these as well.

I'd be happy to work on a pull request.

[image]

foostan, 2024-08-04:
This change will not be accepted, as the cases will no longer be compatible. If I had to do it, I would do it in v5. And if I do it, I want to use the type-c parts the same as others. In that case, the OLED will not be compatible.

closed 2024-08-16 by foostan

PaulRopel, 2025-03-07:
Hi,

I am banging my head arround the firmware, I have also replaced the trrs with usb but do have different pins for rx and tx on both halfs and cannot get the connection to fly, any advice:

https://github.com/PaulRopel/kbd_firmware/blob/main/keyboards/crkbd/qmk/qmk_firmware/rev4_1/config.h

br 
Paul

PaulRopel, 2025-03-09:
> Hi,
>
> I am banging my head arround the firmware, I have also replaced the trrs with usb but do have different pins for rx and tx on both halfs and cannot get the connection to fly, any advice: [...]

all good I was missing the #define SPLIT_USB_DETECT and I realized that I can only assign the pins at compile time not at runtime...