foostan/crkbd #252: Misaligned components on JCLPCB for Corne v4.1

cyuanli, 2024-07-01:
When uploading the production files for Corne v4.1 on JCLPCB, some components appear to be misaligned or rotated the wrong way.

**Cherry**
[image]

**Chocolate**
[image]

Is this an issue on JCLPCB's side or should this be fixed in the files?

foostan, 2024-07-01:
this an issue on JCLPCB's side

Beato99, 2024-07-01:
JLCPCB Engineers will fix it for you before manufacturing, you can upload it like that

cyuanli, 2024-07-05:
I received a message:

> Since we are not so sure about the polarities of the components C5149201 are correct or not. Could you please kindly check if the polarities and placements of the components are correct in the below picture? Is it okay to proceed with production?

[image]

I couldn't find any indications in the documentation, does that seem correct?

foostan, 2024-07-05:
It looks correct. I think this has already been discussed a few times on Discord.

cyuanli, 2024-07-06:
Ah, I didn't think of checking the Discord, will do so in the future. Thanks a lot for the support, it's much appreciated!

closed 2024-08-04 by foostan