foostan/crkbd #254: TRRS and TRS are both available, which one should I choose ? ( & Soldering version estimate?)

NourNuxeo, 2024-07-22:
Hello, (warning: I'm a total beginner)
I was looking to order the PCB + BOM components from JLCPCB and I found the TRRS connector `ST-0277D00-052-142` was back in stock.
In such case should I go for v4 or v4.1 ? It seems to me the main change was the replacement with a TRS connectore because of stock outage.
Just a side question as well: is the soldering version still planned ? (I doubt I'll go for it as I'm a total beginner, I just ask out of curiosity)

Thank you for your time!

foostan, 2024-08-04:
If you use `ST-0277D00-052-142`, you should choose `v4`.

> Just a side question as well: is the soldering version still planned ? (I doubt I'll go for it as I'm a total beginner, I just ask out of curiosity)

I have no plan because the soldering of MCU parts is too difficult to hand solder.

NourNuxeo, 2024-08-05:
In doubt I went with the latest, so 4.1 and the TRS alternative.
Thanks for the answers !

closed 2024-08-05 by NourNuxeo