foostan/crkbd #256: Is it possible at all to add a pointing device to v4?

wzxu, 2024-08-14:
First of all as someone who only learned about the existence of "open source split keyboard" like last week and got hooked, I must say I'm so impressed by foostan's work and all the community's been doing.

Now, I'm still doing my research in hope of building my very first Corne. I like that v4 provides a few more keys because I personally don't see a need of OLED on v3, but I'm kind of confused about the limitation of v4 switching to onboard controller.

Since all the existing builds and components I saw seem to rely on those other controllers people used to need to solder themselves, I'm wondering if it's possible at all for one to build a pointing device into a v4? Namely, even if I can find a way to mount for example a touchpoint to the board, can I even connect it to the controller and will it work?

Much appreciated if anyone and help clarify this or provide some resources I can learn more about. Thanks!

keyboard-magpie, 2024-08-14:
You could add a cirque or azoteq trackpad using the OLED interface (communicating over the I2C protocol). I have made an adapter for this purpose: https://github.com/keyboard-magpie/minimal-fpc-i2c-pcb

The OLED interface is still present on the v4 PCBs.

wzxu, 2024-08-14:
@keyboard-magpie Thanks a lot for sharing. It's good to know. 

I did find [another tutorial for trackpoint](https://github.com/joh/crkbd-trackpoint) and realized the controller mentioned is the same as v4, so I think it's also possible.
The problem is onboard controller chip seems nearly impossible to connect by hand, so will probably need to adjust the PCB design and build with PCBA, and this is way beyond my league.

Compared to that, looks like trackpad will be fairly easy thanks to your adapter! 
Since I'm already used to Mac trackpad, this might even be a better fit for me than trackpoint.
Will do some more research on this. Thanks again!

osbm, 2024-08-14:
Hey @wzxu ,

I am more or less interested in the same type of build. Can you also post pictures here if one day you accomplish that? 
Thanks

closed 2024-10-19 by foostan