foostan/crkbd #257: How to map the extra 4 keys from rev4 using QMK ?

luc-vocab, 2024-08-15:
I am looking at the merge request to add crkbd v4 to QMK here : https://github.com/qmk/qmk_firmware/pull/24084
but it's not clear to me how the extra 4 keys can be mapped, how is it possible using the LAYOUT_split_3x6_3 macro ? Is there a sample somewhere ?

luc-vocab, 2024-08-16:
For the layout macro, we have to use `LAYOUT_split_3x6_3_ex2`
I received a pre-assembled crkbd 4.1 which works fine with the default firmware from https://github.com/foostan/crkbd/blob/main/docs/firmware/rev4/firmware_en.md.
however when I build my own firmware with the above qmk pull request using `qmk compile -kb crkbd/rev4 -km via`, only one half works at a time. Does anyone have any suggestions on troubleshooting steps I could try ?

foostan, 2024-08-16:
Have you checked https://github.com/foostan/kbd_firmware?

luc-vocab, 2024-08-16:
@foostan I am trying right now, will report back very soon

closed 2024-09-20 by foostan