foostan/crkbd #260: PCBA Soldering Issue

joeyweii, 2024-09-05:
After I ordered PCB/PCBA and upload the production files to JLCPCB,
I got the following messages from JLCPCB.

> Soldering issue for some RC8、RC15、C15、RR4 parts  - as advised by the engineer, there will be a soldering issue for the hole in pad design, for instance, skewed parts ect, as below. 

[image]

Is it normal?

marleydepew, 2024-09-13:
What was the result? Is there an error in the files?

marleydepew, 2024-09-16:
Was the problem that there were holes in the pads or that the parts were not aligned to the pads... Or both?

foostan, 2024-09-20:
There is a bit of a risk, to be sure. I will try to fix it someday, but it's not a priority since I've never had a problem with it before.

joeyweii, 2024-09-24:
I have ignored the warning and ordered the PCBs.
The built corne v4 keyboard works very well.
Many thanks for the excellent project!

joeyweii, 2024-09-24:
@black13spell 
I switch the `PCBA Type` to `standard` and the `Assembly Side` to `Bottom Side`.
For other settings, I just used the default.

black13spell, 2024-09-24:
Hi @joeyweii, thanks for the help. You save me of paying extra $40. 

Thanks!!

marleydepew, 2024-09-24:
I paid the extra $0.46 to have them verify the parts placement. Some of them will need rotated or sifted. Some of the parts will have a purple dot in the corner, that usually corresponds to the different shape in the footprint. I looked up each part in the parts catalog to verify the orientation on the footprint

TheRealOne78, 2025-05-04:
Hi,

I've just received the following E-Mail from a JLCPCB AOI staff, regarding V4.1 Cherry:

> Sorry to bother you.
> 
> For the designators R15.RR4.RC15.CED1.RJ1.J1, there are vias on the pads,so the solder would flow into the vias and caused weak soldering after reflowing. Please check the images below.
> 
> [image]
> 
> [image]
> 
> [image]
> 
> May I know if it is OK for you?
> 
> Please consider modifying the design so the soldering can be better in future orders.

I can definitely solder a bit more tin on those pads, but if I wouldn't, this would probably be an issue.

marleydepew, 2025-05-05:
You can change the Vias option to "epoxy filled and capped". That should solve the problem.