foostan/crkbd #261: Q : what is bill of material for v4 ?

sadtab, 2024-09-12:
This is my first attempt to build a keyboard. I'm going with the latest version (v4 as of today).

In the v4 docs : https://github.com/foostan/crkbd/blob/cb4c15542148ac573c55c0e14038ed77c69ca50b/docs/corne-cherry/v4/buildguide_en.md?plain=1#L12

What is PCB : it sounds like an assembled  PCB with all diodes, less, controller, sockets etc. 

In v2 doc, all these components are listed https://github.com/foostan/crkbd/blob/cb4c15542148ac573c55c0e14038ed77c69ca50b/docs/corne-cherry/v2/buildguide_en.md?plain=1#L18

Should I assume BOM of v2 and v4 are the same?

PS : I'm pretty sure github issues is not the right place to ask this question, where is the community forum?  

Peace

jfpcarson, 2024-09-12:
BOM is something new for v4, in previous builds you bought it all and soldered it using the guide. The BOM is located here, for the Corne Chocolate: https://github.com/foostan/crkbd/tree/main/pcbs%2Fcorne-chocolate%2Fhotswap%2Fjlcpcb%2Fproduction_files

You upload that alongside the Gerber files.

Discord is here:
https://discord.gg/6qSGrxq7

marleydepew, 2024-09-13:
I'm thinking there is actually an issue with the BOM or CPL. I'm getting an error when I try to upload the Cherry copy to JLCPCB. I noticed the BOM file lists one kind of MX sockets but the CPL lists three different kinds (1u, 1.5u, 1u/rotary). Not sure if this is causing the error.

[image]

marleydepew, 2024-09-16:
I created a new issue for the errors, since it's a little different from your question.

https://github.com/foostan/crkbd/issues/262

closed 2024-09-16 by sadtab