foostan/crkbd #262: Errors with the CPL

marleydepew, 2024-09-16:
When I upload the cherry BOM and CPL files to JLCPCB, there is an error message that they don't match. There are some differences with the hot swap sockets between the two files but even after removing those I'm still getting the error messages. 

[image]

If you go to the next screen a number of the components are either in the wrong place or have the wrong orientation. For example the USB and TRS jacks are not connected to the PCB.  The components at U2, U3 and U4 are obviously not oriented correctly. I'm not sure if the remaining components are oriented correctly or not.

foostan, 2024-09-19:
I found the cause.
It seems that JLCPCB returns an error if lowercase letters are used in Designator.
For example, if I change `rU1` to `RU1`, no error occurs.
However, even if an error occurs, JLCPCB seems to process it correctly, so it looks like a warning is just being issued. It just seems like a bug.

marleydepew, 2024-09-19:
When you tried it on your end @foostan did the components get placed on the footprints correctly?

foostan, 2024-09-20:
Yes, of course.

marleydepew, 2024-09-22:
Changing the designators to uppercase resolves the error message. The orientation of some parts still needs to be corrected in JLCPCB

closed 2024-09-22 by marleydepew