foostan/crkbd #267: Documentation is really hard for beginners

divad1196, 2024-10-17:
Hi,

First, thank you for the keyboard, it seems really nice.

I have been looking at the documentation and forums for hours but I am not able to understand:
- In the `buildguide_en.md` for V4, I don't see the controller I am supposed to use.
- In the release, with have `pcbs/` and `plates/` folders, but in the `plates/3x6/` folder I also have the `pcb/` folder next to the `svg/`.
  What does that mean ?  If I understand correctly:
  - The SVG files in `svg/`are not used at all (I tried to pass them to my 3D print slicer but I don't think I am supposed to print it from the SVG)
  - I need to pass the GERBER files under `pcb/` to a site like jlcpcb as if it was a PCB, but isn't it really a 3D print ? There are no circuit on it. Is there a reason to make it a PCB ?
  - I am confused with the available files: there is 1 `bottom` file that match all the `top` files ? What is the difference between `cherrymx` and `cherrymx-ex` ? It seems that in the other github's issues, the response is always to use the "cherry" one (the `ex` or the other one?)
  I have a 3D print at home, if possible, I would prefer to print it myself than order it.
- It would be nice to have a link to where we can buy the different components.
- Also (really not that important) why is the `cases` folder missing in the release ?

NOTE: The controller and plate questions are really the main ones. 

I see many of these questions in the issues, but I am still not able to understand what needs to be done.
Maybe putting these in a FAQ file would make it easier?

Sorry for all these questions, and I understand that many people here don't need these answers as they are already experienced, but any help would be really appreciated.

Have a nice day.

waffle87, 2024-10-23:
> In the `buildguide_en.md` for V4, I don't see the controller I am supposed to use.

This is because v4 uses an integrated microcontroller (RP2040). This is not something most users solder themselves and I would say is out of the scope of the build guide — it is mostly for general assembly.

If you're looking for a DIY keyboard (soldering a development board style microcontroller like a Pro Micro, and other components like diodes), v4 probably is not right for you — you should look at v3 PCBs.

> The SVG files in svg/are not used at all (I tried to pass them to my 3D print slicer but I don't think I am supposed to print it from the SVG)

Correct. SVG files are not 3D printable — you would need to convert to them to STL files that your slicer understands.

> I need to pass the GERBER files under pcb/ to a site like jlcpcb as if it was a PCB, but isn't it really a 3D print ? There are no circuit on it. Is there a reason to make it a PCB ? I guess that the pcb works for both left and right?

It is not a 3D print. The plates under `crkbd-4.1/plates/3x6/pcb` are if you would like an FR4 plate — it is the same material PCBs are made of and a pretty decent keyboard plate material.

> there is 1 bottom file that match all the top files ? 

The `plate-bottom.*` file is a back plate for the keyboard. (Google "FR4 sandwich keyboard" if this doesn't mean anything to you)

> What is the difference between cherrymx and cherrymx-ex ?

`plate-top-cherrymx-ex.*` is if you would like the extra two keys you can optionally install on the v4 PCBs.

`plate-top-cherrymx.*` does not contain the extra keys, and is only the 3x6+3 layout.

> It would be nice to have a link to where we can buy the different components.

Assuming you're referring to v4 PCBs, most of the components are electronic components that you will likely source from the PCB manufacturer for them to be assembled with the PCBs.

divad1196, 2024-10-23:
Hi @waffle87 

Thank you very much for responding in details and to all my questions. I was already sorry for asking newbie questions but I realize I was completely off..

Understood, there are no additional control as it is already on the PCB. I am not looking for anything special, I just thought that the controller had to be added manually afterward.

I am wondering: is there any downside to using the RP2040? Like maybe the ease during development? The price difference is so huge. 

I didn't find much on "FR4 samdwich keyboard", but I think I understand the idea from the Corne's pictures. 

The "-ex" is the version with the knobs? Or is it possible to put additional keys?

For the last part: at the time I wrote my questions, I already found out the BoM and PCL files. I was refering to things likr the M2 spacers.

About the BoM and PCL files:
- when I upload them, I always get a warning that they don't match
- the hotswap connector for hailh choc isn't available anymore and they are not able to obtain a quote from their provider so they refunded me. It does not seem to be any valid alternative on jlcpcb. I am now trying it with the MX version.

foostan, 2024-11-15:
> I am wondering: is there any downside to using the RP2040? Like maybe the ease during development? The price difference is so huge.

I have discontinued ProMicro because it can be simplified and compacted. The changes in v4 are available at https://github.com/foostan/crkbd/releases/tag/v4.0.

The RP2040 chip is cheaper than the one used in ProMicro. The PCBA process makes it more expensive. However, it also becomes cheaper if you make a large number of them. It is true that it is more expensive if you make a small number of them. I would like to find a way to make them cheaper, even in small quantities, at some point.

> The "-ex" is the version with the knobs? Or is it possible to put additional keys?

It's for both versions.
Please confirm the data.

> when I upload them, I always get a warning that they don't match

It's the same as https://github.com/foostan/crkbd/issues/272 and you can ignore it.

> the hotswap connector for hailh choc isn't available anymore and they are not able to obtain a quote from their provider so they refunded me. It does not seem to be any valid alternative on jlcpcb. I am now trying it with the MX version.

It's correct. I don't suggest using JLCPCB now; they do not have hot-swap parts.

zsoltkacsandi, 2025-02-21:
> It's correct. I don't suggest using JLCPCB now; they do not have hot-swap parts.

Do you have any other suggestions what PCB assembly service should we use?

I am a beginner and trying to build a Corne keyboard but I am having a hard time what should I do exactly and with which files.
If you can help, I am more than happy to contribute back into the documentation.

divad1196, 2025-02-22:
Hi @zsoltkacsandi ,
There are other companies like [PCBWay](https://www.pcbway.com/) . The files you get from this repository would work in many of these platforms. They will also provide you with alternatives (Sometimes).

I was finally able to get my keyboard done by JLCPCB. I used the connector from another brand that was also waaay cheaper. But honestly, beware the costs. You cannot ask for less than 5 PCB, you cannot order the components from them without getting it assembled for you. (But you can ask to assemble only one PCB). They made a mistake in my order and I got invoiced a lot more than expected, and I got charged additional customs by DHL because it was made in China.

I also personally struggle to find the components I needed (especially M2 spacers and screws, TRRS cable, ...) 
The whole thing costed me around 250CHF. To that, I need to add the switches and keycaps that are really expensive (I 3D printed the case). Now, I need to find what to do with the 4 other assemble PCB, probably finish them and try to sell them.

Good luck.

closed 2025-02-22 by divad1196