foostan/crkbd #268: Is it possible to use ProMicro NRF52840 or ProMicro RP2040?

angelosne, 2024-10-21:
Hello,
I bought those two proMircro alternatives knockoffs from aliexpress.

Is it possible to make it work with this keyboard? If yes what is the workflow I should follow to have a working firmware?

Sorry if question is trivial but I am quite new.

keyboard-magpie, 2024-10-21:
They will probably work, the nrf52840 with ZMK firmware, and the pro micro rp2040 with QMK firmware.

Getting started with those firmwares is well documented, both firmwares have excellent documentation websites - https://zmk.dev/ and https://docs.qmk.fm/.

Using issues is not the best for new user support nowadays- conversational support is much easier on Discord as mentioned above. https://discord.gg/A2DN7468 for ZMK, https://discord.gg/s4gDDPUT for QMK.

angelosne, 2024-10-21:
Thank you very much I will read those!

closed 2024-10-21 by angelosne