foostan/crkbd #271: kailh CPG151101S11-16 & kailh CPG135001S30 unavailable. lternatives

divad1196, 2024-10-27:
Hi,

I tried to place order for the Choc version then the MX version.
In both cases, the "Mechanical Keyboard Shaft ROHS" was not available in sufficient quantities and I had to pre-order them.

The issue is: they are unable to re-order them.
[image]

I found www.lcsc.com that looks similar to the shop in JLCPCB and found this component:
[HanElectricity CPG151101S11-16 ](https://www.lcsc.com/product-detail/Mechanical-Keyboard-Shaft_HanElectricity-CPG151101S11-16_C41430893.html) that is similar to [Kailh CPG151101S11-16](https://www.lcsc.com/product-detail/Mechanical-Keyboard-Shaft_Kailh-CPG151101S11-16_C5156480.html) but available and a almost 5 times cheaper. It also appears in [JLCPCB' shop but without any price](https://jlcpcb.com/partdetail/Hanelectricity-CPG151101S1116/C41430893).

Would it be a valid alternative, or is there any alternative product I could use?

foostan, 2024-10-27:
I saw [HanElectricity CPG151101S11-16](https://www.lcsc.com/product-detail/Mechanical-Keyboard-Shaft_HanElectricity-CPG151101S11-16_C41430893.html) for the first time, but looks compatible. Also, you can buy from other shops such as [kbdfans](https://kbdfans.com/products/mechanical-keyboard-switches-kailh-pcb-socket?variant=39678399381643), other keyboard vendor shops, aliexpress and more.

dahmwern, 2024-11-01:
They're available on Amazon. I just bought 50 here: https://www.amazon.com/dp/B0BVH6M5FP?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1

divad1196, 2024-11-02:
@dahmwern they are way more expensive than what they were on jlcpcb, which is also way more expensive than the alternative here..

I just received the following question from JLCPCB:
> Since we are not so sure about the polarities of [C5149201](https://jlcpcb.com/partdetail/OpscoOptoelectronics-SK6812MINIE/C5149201) are correct or not. Could you please kindly check if the polarities and placements of the components are correct in the attached picture?
> [image]

I don't know anything about the polarity.. but the placement seems to match the shapes and intended final result.

NOTE: The piece [C5149201](https://jlcpcb.com/partdetail/OpscoOptoelectronics-SK6812MINIE/C5149201) is the rgb led.

dahmwern, 2024-11-02:
That's the correct polarity. You can check polarity in KiCAD (free) using the files provided by Foostan on git.

foostan, 2024-11-15:
As @dahmwern says, it looks correct.

Chell-M, 2024-12-05:
@divad1196 did you proceed with the alternative HanElectricity CPG151101S11-16? 

if so can you confirm they indeed work as needed?

divad1196, 2024-12-06:
> @divad1196 did you proceed with the alternative HanElectricity CPG151101S11-16? 
> 
> if so can you confirm they indeed work as needed?

"As needed" I guess yes.
At least, my keys are all responding.

But I am not actively using it for various reasons, so I cannot tell if they are good or not and if they will break in a few weeks.

foostan, 2025-02-09:
This issue looks to be resolved so I will close it.

closed 2025-02-09 by foostan