foostan/crkbd #272: error: The BOM file doesn't match the CPL file, please check and re-upload the files.

CorbinWunderlich, 2024-10-30:
[image]

This happens when I try to process the CPL and BOM. I'm using the files in the production_files directory.

PaulRopel, 2024-10-31:
same issue here.

divad1196, 2024-11-02:
@PaulRopel  @CorbinWunderlich I had the same issue, but if you try to proceed with "Continue" and check, everything should be in order. I guess there is an issue with how the header is read by the website.
 
But I don't know why, the PCB Assembly was said to be $25 during all the steps, but I got invoice like $111 for the final assembly (not including delivery or the PCB itself)... so be cautious

PaulRopel, 2024-11-09:
@divad1196 i cannot continue...

divad1196, 2024-11-09:
> @divad1196 i cannot continue...

Could you help us help you? Just saying "doesn't work" isn't useful. Can you put a screenshot to elaborate what you get?

Yesterday there was a lot of bug on their website "server error". So this might be what you mean now and you just meeded to wait.

Sometimes I would get an upload error.
You just retry and you should have the pop up that @CorbinWunderlich gave. If you look at the screenshot, it has the "continue" button. 
The issue was caused because JLCPCB was trying to use the header as a component (again, a bug) and yesterday evening, it seemed like they fixed that (i.e. I don't get the pop-up)

TL;DR: put a screenshot..

foostan, 2024-11-15:
This issue looks the same as https://discord.com/channels/715899960327012423/715932614598328340/1286343414161477682
I think this is a bug of JLCPCB, so you can ignore it.

foostan, 2025-02-09:
This issue looks to be resolved so I will close it.

closed 2025-02-09 by foostan