foostan/crkbd #273: RGB Matrix Lighting Control Issues in Firmware

dahmwern, 2024-11-01:
I'm having an issue being able to control the RGB Lighting in my keymap.c / config.h / rules.mk files. I've read through all the variants of what's offered in the QMK Guide for the Corne v4.1 Cherry Standard, but nothing seems to be able to control any lighting effects.  

**I've tried:**
`BACKLIGHT_ENABLE = yes`
This doesn't work, it doesn't do anything

`RGBLIGHT_ENABLE = yes`
Which gives me a warning during compiling that statas that RGB_MATRIX_DRIVER is set to WS2812 already and I shouldn't activate both simultaneously. 

 Using both of these, I've applied various cominations of the corresponding keycodes into my keymap to no avail.

**I need to try:**
Adding the following to my `rules.mk`:
`RGB_MATRIX_DRIVER = WS2812` OR `LED_MATRIX_DRIVER = ???`

Does anyone know how this works? I could really use the help.

foostan, 2024-11-15:
Why do we need to set BACKLIGHT_ENABLE = yes?
Is there any issue with https://github.com/foostan/kbd_firmware/blob/main/keyboards/crkbd/qmk/qmk_firmware/rev4_1/info.json#L17 ?

foostan, 2025-02-09:
No reply so I'll close it.

closed 2025-02-09 by foostan