foostan/crkbd #275: < 0.1mm track spacing on v4.1 choc

petermarks, 2024-11-14:
I'm trying to get some boards manufactured by PCBWay. The uploaded gerber fails their checks. They say it has track spacing less than 0.1mm and provide the following image:

[image]

Are you aware of this, or any other issues when uploading to PCBWay?

I'll try to fix it over the weekend and will submit a patch if you like.

foostan, 2024-11-15:
Thank you for sharing. You can fix it, but connecting it to the circuit is no problem. It may be possible to manufacture it as is.

gr8h, 2025-02-19:
@petermarks did you manage to fix this?

petermarks, 2025-02-19:
I had a quick look, but I'm not a kicad user, so didn't get anywhere without some investment in learning the tool. JLBPCB doesn't complain about it and the boards I had made by them work perfectly. As @foostan says, it isn't a problem as both tracks could be connected.

foostan, 2026-11-14:
Thanks for reporting this. In v5 (#328), the PCB has been re-routed from scratch with a minimum clearance of 0.19 mm, and DRC passes with no errors. The production files for v5 should pass the standard checks at PCBWay as well as JLCPCB.

For v4.1, as mentioned above, the affected tracks are on the same net, so the boards work as is.

I'll close this issue when v5 is released.