foostan/crkbd #280: V4 Bluetooth pcb.

pagedpenguin251, 2025-01-02:
This isn't really an issue but i designed a pcb for a nice!nano and nice!view as a mix between the v4 and v3 pcbs. This is me asking if you would be interested in this (I don't want to figure out where all my libraries are unless it's needed)
[image]

foostan, 2025-01-03:
Thank you for sharing your idea!

However nice!nano isn't used in Japan under the Japanese law 😢 
Also, this is not a problem, so could you share ideas and discussions on discord?

pagedpenguin251, 2025-01-03:
I did not know there was a discord, and cant seem to find it.

pagedpenguin251, 2025-01-03:
nevermind I found it, will continue there.

closed 2025-01-03 by pagedpenguin251