foostan/crkbd #281: Can't find the top stl of chocolate version

dty2, 2025-01-16:
As the title suggests, I can't find the top stl of chocolate verion.
I've tried really hard to find it. Could anyone tell me where is it, please？
I find a "top stl" in "pcbs/common/kbd/kicad-packages3D/kbd.3dshapes" called "corne-plate-acrylic-top-left-no-holes.step"
but it seems that the wrong size.

kijewskimateusz, 2025-01-25:
Attaching step files. Generated them by exporting `kicad_pcb` files using pcbnew application. Cherry ones march choc profile

[cherrymx-top-plates.zip](https://github.com/user-attachments/files/18547298/cherrymx-top-plates.zip)

foostan, 2025-01-26:
the chocolate version is the same as the cherry version.

dty2, 2025-01-27:
thank上 for reply, I succeeded in making it
@foostan @kijewskimateusz

closed 2025-01-27 by dty2