foostan/crkbd #282: Keyboard not working properly (one side is off with jack cable)

AndresMarronP, 2025-01-17:
Hello,

I purchased this keyboard three weeks ago, and I’ve recently encountered an issue with the interconnection cable. **One side of the keyboard stops working when connected via the double jack cable.**

I’ve tested the double jack cable with my headphones, and it seems to work fine, so I find it strange that this issue has arisen so soon after purchase. Also, the keyboard is working fine if we connected each part with a different USB Cable.

I hope this is just a software bug or a minor issue that can be resolved, as I’m quite disappointed. Additionally, I noticed another problem with the phone compatibility, which I wasn’t informed about at the time of purchase.

I’d appreciate your assistance in resolving this matter.

foostan, 2025-01-26:
I have not tested the operation with a double jack cable or stated that it is compatible. If other people also require a double jack cable, I will prioritize investigating it, but at this stage, I am not particularly considering it.

AndresMarronP, 2025-01-30:
[image]

I am referring to this middle cable (the one that connects both parts). As you can see, the other side is not working properly through this cable.

Jufralice, 2025-02-04:
Hello, 

I have the same issue.
I have this keyboards since a few weeks now, and two days ago when I plugged in the keyboard into my computer, the two haves were not communicating anymore (just like explained @AndresMarronP ).
I did not flash any of the two halved recently, so I suspect that it's more an hardware issue than software...
Same as for @AndresMarronP, the two part works independently. But apparently no data pass through the serial cable anymore.
I also tested with another cable and that doesn't fix the problem.
My next step will be to disassemble the keyboard and check the welding of the jack connectors I guess...
then I'll try to test if a signal pass trough the jack connector using a voltmeter... somehow... 
Any advice on how I should test/investigate this problem?

Thank you.

foostan, 2025-02-09:
> My next step will be to disassemble the keyboard and check the welding of the jack connectors I guess...
then I'll try to test if a signal pass trough the jack connector using a voltmeter... somehow...

@AndresMarronP @Jufralice 

I think this approach should be good.
Could you tell me your version? 4.1?

If you have just purchased the item, please contact the store. There is a possibility that the item is defective.

Jufralice, 2025-02-09:
Hello, 
Thank you for your response.

Yes I use the rev4_1/standard version

I guess that as it worked and then suddenly stopped working the day after without having flash it in the meantime, we can conclude that it's an hardware issue and not software?

Thanks.

Jufralice, 2025-02-09:
I took pictures of the jack connectors on the two parts. 
I have the impression that some soldering might be a bit weak but I think that the contact should be done anyway... 
But I'm not an expert at all... do you maybe see obvious soldering issues?

Thanks a lot.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jufralice, 2025-02-16:
Hello,

Here are the following on my investigations

I tested the TRS cable connections using a voltmeter.
The connections seems ok:
- I mesure no resistance between each 3 pairs (tip to tip, ring to ring and slave to slave)
- I mesure infinite resistance (no short) between all other combinations (tip to ring, tip to slave, ring to slave,....)

I then measured the voltage when the keyboard is plugged-in (master half connected to the computer and the two halves connected to each other with the TRS cable): 
- I mesure 3.3V between the ground and the ring of the TRS => even when I mesure it from the ground of the master half to the ring of the slave half, or the other way around
- I mesure 5V between the ground and the tip of the TRS => even when I mesure it from the ground of the master half to the tip of the slave half, or the other way around
- and I mesure 3.3V on all the switches (keys) (between the ground and the other switch pin) => Even on the slave half of the keyboard (the one that is not connected to the computer).

I then conclude that the 'power' do go through the slave half via the TRS cable
But for some reasons the 'signals' (for the leds and on keypresses) are not going through.... 

I guess that I am supposed to mesure a voltage variation between the ground and the ring or the tip of the TRS cable when I press a key right?
Because I do not mesure any voltage variation at all...

When I run the 'debug' mode I get these logs:
foostan:Corne v4:1: SPLIT: receiving handshake failed
foostan:Corne v4:1: Failed to execute slave_matrix
foostan:Corne v4:1: Target disconnected, throttling connection attempts

Do you confirm that it's normal that we have 3.3V on the ring of the TRS and 5V on the tip of the TRS?

Do you have any idea of what could be the problem? Could it still be a soldering issue even though I get these voltages?

Any idea of what I could investigate next ? 
Or should I stop the investigation and return it to the shop?

Thanks for you help

foostan, 2025-02-17:
Thank you for confirming.

> Do you confirm that it's normal that we have 3.3V on the ring of the TRS and 5V on the tip of the TRS

It’s normal.
This is probably a hardware problem. The path from the TRS to the RP2040 may be broken or the RP2040 may be broken. It seems that it may fail if the TRS cable is unplugged and plugged in while the power is on.

> Any idea of what I could investigate next ?
Or should I stop the investigation and return it to the shop?

If you have another pair you can determine which one is broken. I am not a seller, so please decide whether or not to contact the shop.

closed 2025-03-28 by foostan

Hlanden, 2025-06-11:
Hi, 

I have the exact same issue as described by @Jufralice above. It most likely happened because I unplugged the TRS when the USB was powered on. 

My investigations also shows that both LDOs seems to work just fine. I have stable 3.3 V and 5 V. 
Both boards work when connected directly to USB.

So that leaves the TX line between the two RP2040s. Is this a correct assumption? Do you have any suggestions on how to fix this, or is it the internals of the MCU that are fried for the TX pin...? @foostan