foostan/crkbd #284: Manufacture cuestion (resin in vias)

CirKanus, 2025-02-22:
Hi everyone,

I am asking for a price quote to manufacture the PCBs for version 4.1.1, and I have been asked a question that I don’t know how to answer:

PCBWay:
For these holes designed on SMD pads, pls confirm whether these vias shoud be filled with resin for better soldering quality?(The price will be increased) If not, there is a risk of cold joint when soldering the components on board.

If you want us to fill vias with resin, then please confirm if we should process it on only these vias designed on SMD pads or on all the vias? There is no difference in price between these two options.

Is there a real risk of cold joint?

Thanks.

foostan, 2025-02-23:
Did Pcbway explain the problem with an image? I'd like to know where is a weak point.

CirKanus, 2025-02-23:
[image]

foostan, 2025-02-23:
I don't know what pad you're talking about, but you probably don't need to do anything.

closed 2025-02-28 by CirKanus