foostan/crkbd #289: CherryV4 - JLCPBC - HanElectricity CPG151101S11-16 (C41430893) as an alternative to Kailh CPG151101S11-16 (C5156480)

mle98, 2025-03-16:
Hey, I and wanted to get myself the PCBs from jlcpcb and used the GERBER-corne-cherry.zip from the repo. But similar to issue #271, Kailh CPG151101S11-16 (C5156480) is no longer available as a component. 

Since I saw in the other issue that the part from HanElectricity (C41430893) should be compatible, I changed the BOM file to reference it. However, now the orientations are all messed up. 

[image]

I'm very new at this and wanted to ask if it is enough to manually rotate and drag the components in the correct position within jlcpbc's tool, or if this will not be exact enough. This is how it looks like afterward:

[image]

mle98, 2025-03-16:
Looking at the tolerances, I don't think it's a good idea to do this manually by hand.
I think it would be best if the BOM and CPL files could be updated with a component that JLCPBC has on stock.

foostan, 2025-03-16:
If the orientation is wrong, JLCPCB will fix it. I recommend enabling the option to check before manufacturing.

closed 2025-03-28 by foostan