foostan/crkbd #291: Random Disconnection with TRS Cable

MarcioBulla, 2025-03-28:
Hardware Version:
V4.1

Issue Description:
I am experiencing a random issue where the side of my keyboard connected via a TRS cable occasionally freezes. This does not happen frequently but seems to occur after the keyboard has been idle for a long period.

https://github.com/user-attachments/assets/301fcf58-bff8-4efa-9f30-b5b96fed4c35

foostan, 2025-04-02:
Does the problem occur with a different cable?
If you have any problems with your hardware, please contact the place of purchase. It may be a manufacturing defect.

MarcioBulla, 2025-04-02:
Yes, I tried with diferentes type C Cable and TRS cable. Thank Very much for your help!!!

veikolippand, 2025-04-24:
I have same issues. It is noted already in main README that it might be related to EMI.
When my phone is close to the keyboards then it happens quite often.

foostan, 2025-04-26:
The EMI issue is known, and I'm working on it, but showing from the video, it seems to be a different issue, probably a hardware failure.

lusqua, 2025-05-17:
same problem here!

juninhopo, 2025-05-17:
same problem here!

foostan, 2025-05-18:
If you have a problem, please share your current situation in detail. If you suspect a malfunction, please contact the store where you purchased the product first. If there is a design or improvement needed, I will consider it.

MarcioBulla, 2025-05-20:
I already contact the store, but they said that is electromagnetic interference.

foostan, 2025-06-22:
If EMI issue, please move to https://github.com/foostan/crkbd/issues/265

MarcioBulla, 2025-07-16:
This issue appears to be related to EMI. Continuing discussion in #265.

closed 2025-07-16 by MarcioBulla