foostan/crkbd #292: Issue with Corne Chocolate v4 – Only One Side Working

exfinen, 2025-04-09:
I have a Corne Chocolate v4 (pre-assembled), and the left and right halves don’t seem to work together. Only the side connected to the PC is functional—the other side remains unresponsive.

When I connect both halves to the PC using two USB cables, both sides work properly.
I’ve noticed that the non-working side’s LEDs don’t light up at all.

It’s not a TRRS cable issue, since the same cable works perfectly with my Corne Chocolate v3.

I believe the problem started after I connected the left half directly to the PC (without the right half) while trying to update the firmware.

Is there any way to reset the keyboard so that both halves can work together again?

exfinen, 2025-04-09:
I think my case is a bit different -- after connecting the USB cable to the PC, I did not disconnect the TRRS cable i.e. When I connected the left half of the keyboard to the PC, it was not connected to the right half.

foostan, 2025-04-26:
Sorry for the late response. You discovered that the cause was a hardware failure, so can I close this issue? 
I'm sorry for the trouble caused by the failure, and I'll try to improve with a design that will prevent more failures. Thank you for reporting it.

exfinen, 2025-04-26:
Sorry, I should have closed this earlier. It turned out at Yushakobo that both sides were broken. I’m looking forward to seeing improvements in future versions!

closed 2025-04-26 by exfinen