foostan/crkbd #293: only one side rgb work

dty2, 2025-04-10:
the problem is title
I tried flashing the firmware on both the left and right keyboards, but the RGB on whichever side I connected the data cable to worked fine, but not on the other side. However, my key layout was correct and I could type fine on both keyboards.
 and i use qmk compile -kb crkbd/rev4_1/standard -km lucky to generate crkbd_rev4_1_standard_lucky.uf2 file and then mount my 
 keyboard and use mv command to move it to my keyboard
my config.h file is
```
#pragma once

#define SERIAL_USART_TX_PIN GP12
#define USB_VBUS_PIN GP13

#define I2C_DRIVER I2CD1
#define I2C1_SDA_PIN GP6
#define I2C1_SCL_PIN GP7

/* RP2040- and hardware-specific config */
#define RP2040_BOOTLOADER_DOUBLE_TAP_RESET // Activates the double-tap behavior
#define RP2040_BOOTLOADER_DOUBLE_TAP_RESET_TIMEOUT 500U
#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64

#define TAPPING_TERM 180
#define TAPPING_TERM_PER_KEY
#define QUICK_TAP_TERM 90
#define QUICK_TAP_TERM_PER_KEY
#define HOLD_ON_OTHER_KEY_PRESS_PER_KEY

#define EE_HANDS
#define SPLIT_RGB_MATRIX
#define SPLIT_TRANSPORT_MIRROR
#define RGB_MATRIX_SLEEP
#define RGB_MATRIX_LED_PROCESS_LIMIT 46
#define RGB_MATRIX_LED_COUNT 46
```
my keymap.c file is 
```
#include QMK_KEYBOARD_H
#if __has_include("keymap.h")
#    include "keymap.h"
#endif

/* THIS FILE WAS GENERATED!
 *
 * This file was generated by qmk json2c. You may or may not want to
 * edit it directly.
 */

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {[0] = LAYOUT_split_3x6_3_ex2(KC_DEL, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_VOLU, KC_BRIU, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_BSPC, LSFT_T(KC_CAPS), KC_A, KC_S, KC_D, KC_F, KC_G, KC_VOLD, KC_BRID, KC_X, KC_H, KC_J, KC_K, KC_L, RSFT_T(KC_CAPS), KC_LCTL, KC_MINS, KC_EQL, KC_Z, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_RCTL, KC_LALT, LT(3, KC_TAB), LT(1, KC_ESC), LT(1, KC_ENT), KC_SPC, KC_RALT),
                                                              [1] = LAYOUT_split_3x6_3_ex2(KC_DEL, KC_AT, KC_LBRC, KC_RBRC, KC_PIPE, KC_AMPR, KC_NO, KC_NO, KC_PERC, KC_GRV, KC_QUOT, KC_DQUO, KC_TILD, KC_BSPC, KC_COLN, KC_1, KC_2, KC_3, KC_4, KC_5, KC_NO, KC_NO, KC_6, KC_7, KC_8, KC_9, KC_0, KC_SCLN, KC_EXLM, KC_LPRN, KC_RPRN, KC_LCBR, KC_RCBR, KC_BSLS, KC_SLSH, KC_CIRC, KC_ASTR, KC_HASH, KC_DLR, KC_QUES, TG(4), MO(2), KC_TRNS, KC_TRNS, MO(2), TG(4)),
                                                              [2] = LAYOUT_split_3x6_3_ex2(KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_F8, KC_F9, KC_F10, KC_F11, KC_NO, KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F6, KC_NO, KC_NO, KC_F7, KC_LEFT, KC_DOWN, KC_UP, KC_RGHT, KC_F12, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_NO, KC_NO, KC_TRNS, KC_NO),
                                                              [3] = LAYOUT_split_3x6_3_ex2(KC_DEL, KC_Q, KC_W, KC_E, KC_R, KC_NO, KC_NO, KC_NO, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_BSPC, KC_BSPC, KC_1, KC_2, KC_3, KC_4, KC_G, KC_NO, KC_NO, KC_T, KC_5, KC_6, KC_7, KC_8, KC_ENT, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_NO, KC_TRNS, KC_NO, KC_RSFT, KC_TAB, KC_NO),
                                                              [4] = LAYOUT_split_3x6_3_ex2(KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, MO(5), RGB_HUI, RGB_SAI, RGB_SPI, RGB_VAI, KC_NO, KC_NO, KC_NO, KC_NO, RGB_VAD, RGB_SPD, RGB_SAD, RGB_HUD, MO(5), KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, TG(4), RGB_TOG, RGB_RMOD, RGB_MOD, RGB_TOG, TG(4)),
                                                              [5] = LAYOUT_split_3x6_3_ex2(KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_NO, KC_NO, KC_NO, KC_NO, QK_BOOT, KC_NO, KC_NO, QK_BOOT, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO)};

bool rgb_matrix_indicators_user() {
    uint8_t brightness = rgb_matrix_get_val();

    if (layer_state_is(0)) {
        rgb_matrix_set_color_all((0 * brightness) / 255, (47 * brightness) / 255, (167 * brightness) / 255);
    }
    if (layer_state_is(1)) {
        rgb_matrix_set_color_all((200 * brightness) / 255, (150 * brightness) / 255, (255 * brightness) / 255);
    }
    if (layer_state_is(3)) {
        rgb_matrix_set_color_all((1 * brightness) / 255, (132 * brightness) / 255, (127 * brightness) / 255);
    }
    if (host_keyboard_led_state().caps_lock) {
        rgb_matrix_set_color_all((253 * brightness) / 255, (195 * brightness) / 255, (198 * brightness) / 255);
    }
    if (get_mods() & MOD_BIT(KC_LSFT)) {
        rgb_matrix_set_color_all((253 * brightness) / 255, (195 * brightness) / 255, (198 * brightness) / 255);
    }
    if (get_mods() & MOD_BIT(KC_RSFT)) {
        rgb_matrix_set_color_all((253 * brightness) / 255, (195 * brightness) / 255, (198 * brightness) / 255);
    }
    return true;
}

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case LT(3, KC_TAB):
            if (record->tap.count > 0) {
                if (record->event.pressed) {
                    register_code(KC_TAB);
                } else {
                    unregister_code(KC_TAB);
                }
                return false;
            } else {
                if (record->event.pressed) {
                    register_code(KC_LGUI);
                } else {
                    unregister_code(KC_LGUI);
                }
                return true;
            }
        default:
            return true;
    }
}

uint16_t get_tapping_term(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case LSFT_T(KC_CAPS):
            return 999;
        case RSFT_T(KC_CAPS):
            return 999;
        case LT(1, KC_ESC):
            return 999;
        case LT(1, KC_ENT):
            return 999;
        default:
            return TAPPING_TERM;
    }
}

uint16_t get_quick_tap_term(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case LSFT_T(KC_CAPS):
            return 50;
        case RSFT_T(KC_CAPS):
            return 50;
        case LT(1, KC_ESC):
            return 50;
        case LT(1, KC_ENT):
            return 50;
        default:
            return QUICK_TAP_TERM;
    }
}

bool get_hold_on_other_key_press(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case LSFT_T(KC_CAPS):
            return true;
        case RSFT_T(KC_CAPS):
            return true;
        case LT(1, KC_ESC):
            return true;
        case LT(1, KC_ENT):
            return true;
        default:
            return false;
    }
}

#ifdef OTHER_KEYMAP_C
#    include OTHER_KEYMAP_C
#endif // OTHER_KEYMAP_C
```
my rules.mk file is
```
RGBLIGHT_ENABLE = no
RGB_MATRIX_ENABLE = yes
```
and the file tree is:
rules.mk, config.h and keymap.h are all in lucky directory and  lucky directory is in keymaps dir which under the crkbd dir in qmk

foostan, 2025-04-26:
Please let me know what changes you've made to the default config.
It would be best if you shared a git diff with me.

dty2, 2025-04-28:
ok, here is diff

``` bash
 diff ./default/config.h ./lucky/config.h
1,19c1,2
< /*
< This is the c configuration file for the keymap
< 
< Copyright 2012 Jun Wako <wakojun@gmail.com>
< Copyright 2015 Jack Humbert
< 
< This program is free software: you can redistribute it and/or modify
< it under the terms of the GNU General Public License as published by
< the Free Software Foundation, either version 2 of the License, or
< (at your option) any later version.
< 
< This program is distributed in the hope that it will be useful,
< but WITHOUT ANY WARRANTY; without even the implied warranty of
< MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
< GNU General Public License for more details.
< 
< You should have received a copy of the GNU General Public License
< along with this program.  If not, see <http://www.gnu.org/licenses/>.
< */
---
> // Copyright 2024 QMK
> // SPDX-License-Identifier: GPL-2.0-or-later
23,26c6,7
< // #define USE_MATRIX_I2C
< 
< // #define QUICK_TAP_TERM 0
< // #define TAPPING_TERM 100
---
> #define SERIAL_USART_TX_PIN GP12
> #define USB_VBUS_PIN GP13
28,43c9,22
< #ifdef RGBLIGHT_ENABLE
< #    define RGBLIGHT_EFFECT_BREATHING
< #    define RGBLIGHT_EFFECT_RAINBOW_MOOD
< #    define RGBLIGHT_EFFECT_RAINBOW_SWIRL
< #    define RGBLIGHT_EFFECT_SNAKE
< #    define RGBLIGHT_EFFECT_KNIGHT
< #    define RGBLIGHT_EFFECT_CHRISTMAS
< #    define RGBLIGHT_EFFECT_STATIC_GRADIENT
< #    define RGBLIGHT_EFFECT_RGB_TEST
< #    define RGBLIGHT_EFFECT_ALTERNATING
< #    define RGBLIGHT_EFFECT_TWINKLE
< #    define RGBLIGHT_LIMIT_VAL 120
< #    define RGBLIGHT_HUE_STEP 10
< #    define RGBLIGHT_SAT_STEP 17
< #    define RGBLIGHT_VAL_STEP 17
< #endif
---
> #define I2C_DRIVER I2CD1
> #define I2C1_SDA_PIN GP6
> #define I2C1_SCL_PIN GP7
> 
> /* RP2040- and hardware-specific config */
> #define RP2040_BOOTLOADER_DOUBLE_TAP_RESET // Activates the double-tap behavior
> #define RP2040_BOOTLOADER_DOUBLE_TAP_RESET_TIMEOUT 500U
> #define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
> 
> #define TAPPING_TERM 180
> #define TAPPING_TERM_PER_KEY
> #define QUICK_TAP_TERM 90
> #define QUICK_TAP_TERM_PER_KEY
> #define HOLD_ON_OTHER_KEY_PRESS_PER_KEY
```

``` bash
[... 145 lines cut ...]
>             return 50;
>         default:
>             return QUICK_TAP_TERM;
>     }
> }
> 
> bool get_hold_on_other_key_press(uint16_t keycode, keyrecord_t *record) {
>     switch (keycode) {
>         case LSFT_T(KC_CAPS):
>             return true;
>         case RSFT_T(KC_CAPS):
>             return true;
>         case LT(1, KC_ESC):
>             return true;
>         case LT(1, KC_ENT):
>             return true;
>         default:
>             return false;
>     }
> }
> 
> #ifdef OTHER_KEYMAP_C
> #    include OTHER_KEYMAP_C
> #endif // OTHER_KEYMAP_C
```

foostan, 2025-05-10:
```diff
 diff ./default/config.h ./lucky/config.h
1,19c1,2
< /*
< This is the c configuration file for the keymap
< 
< Copyright 2012 Jun Wako <wakojun@gmail.com>
< Copyright 2015 Jack Humbert
< 
< This program is free software: you can redistribute it and/or modify
< it under the terms of the GNU General Public License as published by
< the Free Software Foundation, either version 2 of the License, or
< (at your option) any later version.
< 
< This program is distributed in the hope that it will be useful,
< but WITHOUT ANY WARRANTY; without even the implied warranty of
< MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
< GNU General Public License for more details.
< 
< You should have received a copy of the GNU General Public License
< along with this program.  If not, see <http://www.gnu.org/licenses/>.
< */
---
> // Copyright 2024 QMK
> // SPDX-License-Identifier: GPL-2.0-or-later
23,26c6,7
< // #define USE_MATRIX_I2C
< 
< // #define QUICK_TAP_TERM 0
< // #define TAPPING_TERM 100
---
> #define SERIAL_USART_TX_PIN GP12
> #define USB_VBUS_PIN GP13
28,43c9,22
< #ifdef RGBLIGHT_ENABLE
< #    define RGBLIGHT_EFFECT_BREATHING
< #    define RGBLIGHT_EFFECT_RAINBOW_MOOD
< #    define RGBLIGHT_EFFECT_RAINBOW_SWIRL
< #    define RGBLIGHT_EFFECT_SNAKE
< #    define RGBLIGHT_EFFECT_KNIGHT
< #    define RGBLIGHT_EFFECT_CHRISTMAS
< #    define RGBLIGHT_EFFECT_STATIC_GRADIENT
< #    define RGBLIGHT_EFFECT_RGB_TEST
< #    define RGBLIGHT_EFFECT_ALTERNATING
< #    define RGBLIGHT_EFFECT_TWINKLE
< #    define RGBLIGHT_LIMIT_VAL 120
< #    define RGBLIGHT_HUE_STEP 10
< #    define RGBLIGHT_SAT_STEP 17
< #    define RGBLIGHT_VAL_STEP 17
< #endif
---
> #define I2C_DRIVER I2CD1
> #define I2C1_SDA_PIN GP6
> #define I2C1_SCL_PIN GP7
> 
> /* RP2040- and hardware-specific config */
> #define RP2040_BOOTLOADER_DOUBLE_TAP_RESET // Activates the double-tap behavior
> #define RP2040_BOOTLOADER_DOUBLE_TAP_RESET_TIMEOUT 500U
> #define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
> 
> #define TAPPING_TERM 180
> #define TAPPING_TERM_PER_KEY
> #define QUICK_TAP_TERM 90
> #define QUICK_TAP_TERM_PER_KEY
> #define HOLD_ON_OTHER_KEY_PRESS_PER_KEY
```

``` diff
 diff ./default/keymap.c ./lucky/keymap.c 
1,18d0
[... 143 lines cut ...]
>             return 50;
>         default:
>             return QUICK_TAP_TERM;
>     }
> }
> 
> bool get_hold_on_other_key_press(uint16_t keycode, keyrecord_t *record) {
>     switch (keycode) {
>         case LSFT_T(KC_CAPS):
>             return true;
>         case RSFT_T(KC_CAPS):
>             return true;
>         case LT(1, KC_ESC):
>             return true;
>         case LT(1, KC_ENT):
>             return true;
>         default:
>             return false;
>     }
> }
> 
> #ifdef OTHER_KEYMAP_C
> #    include OTHER_KEYMAP_C
> #endif // OTHER_KEYMAP_C
```