foostan/crkbd #295: Hall Effect PCB

TannerRockCode, 2025-04-15:
I was curious if there is consideration of creating a PCB with Hall Effect compatible with magnetic switches? Super impressed with the project, perhaps I can become a contributor in the long term to help make it happen.

ttelford, 2025-04-20:
I am in _no way_ affiliated with this project, but I had a thought on those lines myself. However, my thought was to do something _very_ different: Rather than using a programmable [Hall Effect _switch_](https://www.digikey.com/en/products/detail/melexis-technologies-nv/MLX92232LSE-AAA-002-RE/5761856), where the sensor detects when the magnetic field is strong enough, and triggers like a normal keyswitch...

Instead, use a [Hall Effect position sensor](https://www.digikey.com/en/products/detail/isentek-inc/iST8801/24389746), which would continually report how far down each key is pressed. This would effectively turn the keyboard into an 'analog' controller with an 'axis' for each key.

The idea that each key would not only be able to have programmable trigger points, but also have a variable key position. For example, in an FPS, the 'wasd' keys could vary the speed of the walk, depending on how far they are pressed. Or you could make it a musical instrument, as it could tell how hard and how fast you press a key (like a piano).

Aside from the challenge of designing a circuit board, you'd also have to design an _entirely_ different kind of firmware. Not quite from scratch, but close.

foostan, 2025-04-26:
I don't have plans to make a HE version at the moment, just because I don't have the skills to do it.
If you understand HE well and know the design trends, I would appreciate it if you could share them with me.

closed 2025-06-22 by foostan