foostan/crkbd #296: Custom Firmware Not Working on Windows

alextverdyy, 2025-04-17:
Hello,

I am experiencing an issue with my Corne v4 keyboard when using custom firmware compiled from the official repository. The firmware works perfectly on macOS and Linux, but it does not function correctly on Windows. Specifically, while the keyboard is detected as a USB device and key presses are logged in QMK Console, no keystrokes are registered in Windows.

## Steps to Reproduce
1. Clone the official repository.
2. Set up the compilation environment following the provided instructions.
3. Compile the firmware for the `crkbd` model using the following commands:
   ```
   make qmk-clean
   kb=crkbd make qmk-init
   kb=crkbd kr=rev4_1/standard km=via make qmk-compile
   ```
4. Flash the compiled firmware onto the keyboard using QMK Toolbox or another method.
5. Test the keyboard on Windows.

## Observed Behavior
- **macOS/Linux**: The keyboard functions correctly with both custom and precompiled firmware.
- **Windows**: The keyboard is detected, but key presses are not registered. Precompiled firmware from the repository works without issues.

## QMK Console Logs
When testing on Windows, the QMK Console logs show that key presses are being received by the firmware. However, these key presses are not being sent to the operating system. Here is an example of the log output:
```
KL: kc: 0x0004, col:  0, row:  0, pressed: 1, time:  1234, int: 0, count: 1
KL: kc: 0x0004, col:  0, row:  0, pressed: 0, time:  1250, int: 0, count: 1
KL: kc: 0x0005, col:  1, row:  0, pressed: 1, time:  1300, int: 0, count: 1
KL: kc: 0x0005, col:  1, row:  0, pressed: 0, time:  1315, int: 0, count: 1
```

## Additional Information
- The problem persists even when using the official configuration files.
- NKRO is disabled (`NKRO_ENABLE = no`).
- OS detection reports the correct host system (`OS_WINDOWS`).
- The keyboard uses the RP2040 processor and the `split_3x6_3` layout.

## Request for Assistance
- Could there be an issue with my compilation process or environment?
- Are there any specific settings required for Windows compatibility that I might be missing?

Thank you for any help or suggestions!

foostan, 2025-04-26:
Let me ask a few additional questions.

- Will firmware compiled on macOS/Linux work on Windows?
- Will firmware compiled on Windows work on macOS/Linux?

Unfortunately, I don't have Windows, so I need help from others to solve this problem. This is the first time I've received a report that it didn't work on Windows.

closed 2025-05-17 by foostan

alextverdyy, 2025-05-17:
Sorry, I have not been able to answer before, I have tried to compile the firmware on both Windows and Mac, but there is no way, in fact the QMK SO Detector detects that it is Windows and debugs the output when it seems to send the keycodes, but I had no way to make it write anything ... It is as if Windows does not recognize it.... I have tried it on 5 different Windows computers and nothing. @foostan

foostan, 2025-05-18:
I don't currently have Windows, so I can't verify this. I've also never heard of the same issues.

I recommend asking around on Discord server of QMK, as there may be people who know more about QMK's behavior.