foostan/crkbd #301: unable to compile for v4.1

ingvardm, 2025-05-08:
I am trying to compile for rev4.1

Following the instructions and running
```kb=crkbd kr=rev4_1/standard km=vial make vial-qmk-compile```

gives this output
```
☒ Not including data from file: keyboards/tmp/crkbd/rev4_1/info.json
☒ 	split.transport.protocol: 'serial_usart' is not one of ['custom', 'i2c', 'serial']
☒ tmp/crkbd/rev4_1/standard: RGB_DISABLE_WHEN_USB_SUSPENDED in config.h is no longer a valid option and should be replaced with RGB_MATRIX_SLEEP
☒ 'matrix_size'
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/qmk/1.1.6/libexec/lib/python3.13/site-packages/milc/milc.py", line 606, in __call__
    return self.__call__()
           ~~~~~~~~~~~~~^^
  File "/opt/homebrew/Cellar/qmk/1.1.6/libexec/lib/python3.13/site-packages/milc/milc.py", line 611, in __call__
    return self._subcommand(self)
           ~~~~~~~~~~~~~~~~^^^^^^
  File "/Users/____/projects/kbd_firmware/src/vial-kb/vial-qmk/lib/python/qmk/cli/generate/rules_mk.py", line 65, in generate_rules_mk
    kb_info_json = dotty(info_json(cli.args.keyboard))
                         ~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "/Users/____/projects/kbd_firmware/src/vial-kb/vial-qmk/lib/python/qmk/info.py", line 257, in info_json
    info_data = _extract_led_config(info_data, str(keyboard))
  File "/Users/____/projects/kbd_firmware/src/vial-kb/vial-qmk/lib/python/qmk/info.py", line 773, in _extract_led_config
    cols = info_data['matrix_size']['cols']
           ~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'matrix_size'

 *                                                                                                     [ERRORS]

builddefs/build_keyboard.mk:235: *** Platform not defined.  Stop.
make: *** [vial-qmk-compile] Error 2
```

What am i doing wrong?

foostan, 2025-05-10:
This issue is the same as https://github.com/foostan/kbd_firmware/issues/13 and already fixed. Please check it again 🙏

ingvardm, 2025-05-12:
Thanks, i already fixed it in the vial-qmk, btw :)

closed 2025-05-17 by foostan