foostan/crkbd #302: JLCPCB - SMT placement

Pauiii, 2025-05-20:
Hello,

I placed a PCB order with JLCPCB yesterday and have since received an email asking me to confirm the component placement. As I am unable to determine whether the components are positioned correctly, I would appreciate it if someone could assist by reviewing and confirming the placement for me.

Here is the original e-mail:

> Hi there,
>
> Sorry to bother you, but there is one thing that we want to confirm with you about your SMT order before proceeding.
>
> Since we are not so sure about the polarities of the components are correct or not. Could you please kindly check if the polarities and placements of the SMD components are correct in the below 2D picture? Is it okay to proceed with production?
>
> Your early reply will be highly appreciated, thank you so much!

[image]

Thank you in advance!

foostan, 2025-06-22:
It looks no problem

closed 2025-07-26 by foostan

igdrakoding90-del, 2026-01-16:
@Pauiii is it working as it should ? i just ordered , and in next 2 days i will get the same email like you...