foostan/crkbd #305: PCD Connection probelm

ivan509-hub, 2025-06-12:
I've just got the PCB boards from JLCPCB, but here're the problems: how to connect left to right, and the connection to the pc? It seems the board reserve the connection port to be soldered with type C connection and the switch, so is it the case? Do I have to solder them to the board? 

[image]

foostan, 2025-06-13:
Please check the BOM file. I don't recommend to implement yourself because it's hard to implement other parts(such as RP2040) manually.

ivan509-hub, 2025-06-13:
Thank you！I have just checked the BOM file and found that there are many components needs to be solder on the board. Is it possible to order them and get them soldered when ordering from JLCPCB or PCBway ( or any other companies)

closed 2025-07-26 by foostan