foostan/crkbd #308: Questions about corne-cherry v2.1 PCB

sahand291, 2025-08-02:
I want to build cherry-corne v2.1. but since I could not found SK6812MINI in Iran market and its hard to buy from outside. Want to change on PCB design to SK6812MINI-ER. and then print. and maybe add a switch since want to make it wireless.

but I have a question. 

1. When i run design rule checker (before changing anything) shows me hundreds of errors. Are these OK? or is just something wrong with rules and they have to be customized to fit with this design?.

2. and is there footprint library of parts? i did not found anything in v2.1 tag. 

[image]

thank you for you great design. :)

foostan, 2025-08-02:
Most errors should be fine, but since it's old data, some problems may remain. One reason is that the Kicad version has been updated and the number of design checks has increased.

Please search footprint data from https://github.com/foostan/kbd. v2 parts may have changed now.

sahand291, 2025-08-02:
thanks for the respond. i will change the led to SK6812-ER and try to add battery socket and switch. hope it work. and will share about my experience in discord server :) 

I'm closing this issue.

closed 2025-08-02 by sahand291